esphome/feature-requests #1062: Oral B toothbrush BLE implementation integration: esp32_ble_tracker

andres-a-git, 2020-12-30:
**Describe the problem you have/What new integration you would like**

I would like to have the Oral B toothbrush data over bluetooth. 

**Please describe your use case for this integration and alternatives you've tried:**

Use case: pull brushing data (as it is already possible with Xiaomi MiFlora sensor) and display it in Home Assistant and also in TTGO screens.

**Additional context**

Somebody has already prepared some code that might be copied or pulled:
https://github.com/imduffy15/esphome/commit/90c3cb62b37e9ce308ed0c9d510f54ff95c2dfec

iMicknl, 2021-07-05:
@andres-a-git did you already make a start for this? I was looking into this a while ago and perhaps I will give it a try again soon :-).

andres-a-git, 2021-07-06:
No movement by me because of the lack of knowledge

NODeeJay, 2021-12-05:
I digged into it and tried to test it with esphome custom/external/my components feature. There seem to be a way with local files and git. It seems that git also supports storing them in the folder /esphome/components - in this case the import is straight forward, while the local way seem to only allow storing them under /esphome/my_components - which means one the files need to adapted.
How can I activate debug logging for ESPhome in home assistant? Under the add-on I did not find a setting and home assistant seems to ignore my logging setting.

andres-a-git, 2022-02-15:
> After some great help from @ssieb on discord it came up and to ESP and transferred to HA.
> 
> To get it running (quick and dirty under Home Assistant supervised):
> 
> 
> 
> - in your preferred config editor under _esphome_ create a folder called _components_
> 
> - copy the folders _oralb_ble_ and _oralb_brush_ and their content from https://github.com/imduffy15/esphome/commit/90c3cb62b37e9ce308ed0c9d510f54ff95c2dfec to _/esphome/components_
> 
> - update your esp yaml with:
> 
> ```
> 
> esp32_ble_tracker:
> 
> external_components:
> 
>   - source:
> 
>       type: local
> 
>       path: components
> 
>     components: [oralb_ble,oralb_brush]
> 
> oralb_ble:
> 
> sensor:
> 
>   - platform: oralb_brush
> 
>     mac_address: XX:XX:XX:XX:XX:XX
> 
>     state:
> 
>       name: "OralB State"
> 
> ```
> 
> 
> 
> - Install and enjoy the log sending state 2 or 3
> 
> - Important! BLE tracker needs to re-partition the ESP, when you activate BLE tracker the first time. You need to connect the ESP this one time over USB, afterwards OTA can be used again.
> 
> 
> 
> I need some help getting the states implemented that the brush can show correctly what it does, see https://github.com/zewelor/bt-mqtt-gateway/blob/master/workers/toothbrush_homeassistant.py
> 
> 
> 
> --------- earlier that evening -------
> 
> I digged into it and tried to test it with esphome custom/external/my components feature. There seem to be a way with local files and git. It seems that git also supports storing them in the folder /esphome/components - in this case the import is straight forward, while the local way seem to only allow storing them under /esphome/my_components - which means one the files need to adapted.
> 
> How can I activate debug logging for ESPhome in home assistant? Under the add-on I did not find a setting and home assistant seems to ignore my logging setting.

Well, I have now data flowing into HA. Thanks!

closed 2022-07-01 by nagyrobi

nstrelow, 2022-07-10:
When I run this code, I get an error:
```
File "...external_components\65b48a14\esphome\components\oralb_brush\sensor.py", line 23, in <module>
    cv.Optional(CONF_STATE): sensor.sensor_schema(UNIT_EMPTY, ICON_EMPTY, 0),
TypeError: sensor_schema() takes from 0 to 1 positional arguments but 3 were given
```

Did the esphome interface change? I might clone the repo and make adjustments

ssieb, 2022-07-11:
Yes, it was switched to use named arguments instead.  If you're going to make those adjustments, you could make a PR for esphome as well to add this.

nstrelow, 2022-07-14:
I am using this now: https://community.home-assistant.io/t/oralb-toothbrush-with-esphome-ble/377471
Works good enough for my case :)