esphome/feature-requests #1064: Dallas on more then one Pin

MiG-41, 2020-12-31:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

lwqcz, 2021-01-07:
Hi, did you read something about 1wire bus design? My 1w is capable of handling 18x DS18B20 with an ease...

MiG-41, 2021-01-08:
Good for you :) 
Using orginal Maxim DS18B20 schould gives such numbers , unfortunatly on market there is more cheap fakes... :https://github.com/cpetrich/counterfeit_DS18B20
Unfortunatly i have a lot of noisy shit , with now i can't replace ( they are under a concrete) and have to use them as they are. To late for design.

So let's focus why ESPHome can detect 1wires on more than one pin (bus) , but only from last in the list is able to read later....

MiG-41, 2021-01-22:
Additional example. 
In config 1Pcs of Dallas on GPIO4 ( listed in config first) , and on GPIO0 4Pcs (listed in config last). 
Not possible to read temperature from any of connected to GPIO0.
`
INFO Reading configuration /config/esphome/dallas2.yaml...
INFO Starting log output from dallas2.local using esphome API
INFO Connecting to dallas2.local:6053 (192.168.1.148)
INFO Successfully connected to dallas2.local
[17:02:29][I][app:105]: ESPHome version 1.15.3 compiled on Jan 22 2021, 16:56:30
[17:02:29][C][wifi:443]: WiFi:
[17:02:29][C][wifi:303]:   SSID: [redacted]
[17:02:29][C][wifi:304]:   IP Address: 192.168.1.148
[17:02:29][C][wifi:306]:   BSSID: [redacted]
[17:02:29][C][wifi:307]:   Hostname: 'dallas2'
[17:02:29][C][wifi:311]:   Signal strength: -60 dB ▂▄▆█
[17:02:29][C][wifi:315]:   Channel: 5
[17:02:29][C][wifi:316]:   Subnet: 255.255.255.0
[17:02:29][C][wifi:317]:   Gateway: 192.168.1.1
[17:02:29][C][wifi:318]:   DNS1: 8.8.8.8
[17:02:29][C][wifi:319]:   DNS2: 8.8.4.4
[17:02:29][C][logger:185]: Logger:
[17:02:29][C][logger:186]:   Level: DEBUG
[17:02:29][C][logger:187]:   Log Baud Rate: 115200
[17:02:29][C][logger:188]:   Hardware UART: UART0
[17:02:29][C][dallas.sensor:072]: DallasComponent:
[17:02:29][C][dallas.sensor:073]:   Pin: GPIO4 (Mode: INPUT)
[17:02:29][C][dallas.sensor:074]:   Update Interval: 20.0s
[17:02:29][D][dallas.sensor:079]:   Found sensors:
[17:02:29][D][dallas.sensor:082]:     0xB20300A279AB5828
[17:02:29][C][dallas.sensor:087]:   Device 'Z1 Temperature'
[17:02:29][C][dallas.sensor:087]:     Unit of Measurement: '°C'
[17:02:29][C][dallas.sensor:087]:     Accuracy Decimals: 1
[17:02:29][C][dallas.sensor:087]:     Icon: 'mdi:thermometer'
[17:02:29][C][dallas.sensor:095]:     Address: 0xB20300A279AB5828
[17:02:29][C][dallas.sensor:096]:     Resolution: 12
[17:02:29][C][dallas.sensor:087]:   Device 'Z2 Temperature'
[17:02:29][C][dallas.sensor:087]:     Unit of Measurement: '°C'
[17:02:29][C][dallas.sensor:087]:     Accuracy Decimals: 1
[17:02:29][C][dallas.sensor:087]:     Icon: 'mdi:thermometer'
[17:02:29][C][dallas.sensor:095]:     Address: 0x5002119177B7B228
[17:02:29][C][dallas.sensor:096]:     Resolution: 12
[17:02:29][C][dallas.sensor:087]:   Device 'Z3 Temperature'
[17:02:29][C][dallas.sensor:087]:     Unit of Measurement: '°C'
[17:02:29][C][dallas.sensor:087]:     Accuracy Decimals: 1
[17:02:29][C][dallas.sensor:087]:     Icon: 'mdi:thermometer'
[17:02:29][C][dallas.sensor:095]:     Address: 0xC6021391771EF428
[17:02:29][C][dallas.sensor:096]:     Resolution: 12
[17:02:29][C][dallas.sensor:072]: DallasComponent:
[17:02:29][C][dallas.sensor:073]:   Pin: GPIO0 (Mode: INPUT)
[17:02:29][C][dallas.sensor:074]:   Update Interval: 20.0s
[17:02:29][D][dallas.sensor:079]:   Found sensors:
[17:02:29][D][dallas.sensor:082]:     0x6A021791772CC828
[17:02:29][D][dallas.sensor:082]:     0x7702159246EFA828
[17:02:29][D][dallas.sensor:082]:     0xC6021391771EF428
[17:02:29][D][dallas.sensor:082]:     0x5002119177B7B228
[17:02:29][C][captive_portal:169]: Captive Portal:
[17:02:29][C][ota:029]: Over-The-Air Updates:
[17:02:29][C][ota:030]:   Address: dallas2.local:8266
[17:02:29][C][ota:032]:   Using Password.
[17:02:29][C][api:095]: API Server:
[17:02:29][C][api:096]:   Address: dallas2.local:6053
[17:02:29][C][mqtt:051]: MQTT:
[17:02:29][C][mqtt:053]:   Server Address: 192.168.1.90:1883 (192.168.1.90)
[17:02:29][C][mqtt:054]:   Username: [redacted]
[17:02:29][C][mqtt:055]:   Client ID: [redacted]
[17:02:29][C][mqtt:057]:   Discovery prefix: 'homeassistant'
[17:02:29][C][mqtt:058]:   Discovery retain: YES
[17:02:29][C][mqtt:060]:   Topic Prefix: 'dallas2'
[17:02:29][C][mqtt:062]:   Log Topic: 'dallas2/debug'
[17:02:29][C][mqtt:065]:   Availability: 'dallas2/status'
[... 66 lines cut ...]
[17:08:19][C][ota:032]:   Using Password.
[17:08:19][C][api:095]: API Server:
[17:08:19][C][api:096]:   Address: dallas2.local:6053
[17:08:19][C][mqtt:051]: MQTT:
[17:08:19][C][mqtt:053]:   Server Address: 192.168.1.90:1883 (192.168.1.90)
[17:08:19][C][mqtt:054]:   Username: [redacted]
[17:08:19][C][mqtt:055]:   Client ID: [redacted]
[17:08:19][C][mqtt:057]:   Discovery prefix: 'homeassistant'
[17:08:19][C][mqtt:058]:   Discovery retain: YES
[17:08:19][C][mqtt:060]:   Topic Prefix: 'dallas2'
[17:08:19][C][mqtt:062]:   Log Topic: 'dallas2/debug'
[17:08:19][C][mqtt:065]:   Availability: 'dallas2/status'
[17:08:19][C][mqtt.sensor:024]: MQTT Sensor 'Z1 Temperature':
[17:08:19][C][mqtt.sensor:028]:   State Topic: 'dallas2/sensor/z1_temperature/state'
[17:08:19][C][mqtt.sensor:024]: MQTT Sensor 'Z2 Temperature':
[17:08:19][C][mqtt.sensor:028]:   State Topic: 'dallas2/sensor/z2_temperature/state'
[17:08:19][C][mqtt.sensor:024]: MQTT Sensor 'Z3 Temperature':
[17:08:19][C][mqtt.sensor:028]:   State Topic: 'dallas2/sensor/z3_temperature/state'
[17:08:24][W][dallas.sensor:146]: 'Z1 Temperature' - Scratch pad checksum invalid!
[17:08:24][D][sensor:092]: 'Z1 Temperature': Sending state nan °C with 1 decimals of accuracy
[17:08:24][D][dallas.sensor:153]: 'Z2 Temperature': Got Temperature=22.6°C
[17:08:24][D][sensor:092]: 'Z2 Temperature': Sending state 22.56250 °C with 1 decimals of accuracy
[17:08:24][D][dallas.sensor:153]: 'Z3 Temperature': Got Temperature=23.0°C
[17:08:24][D][sensor:092]: 'Z3 Temperature': Sending state 23.00000 °C with 1 decimals of accuracy
`

MiG-41, 2023-02-26:
A lot of time passed , and tried once again to compile , but this time , when there is more then one gpio configured in dallas section ,during compilation error imidietly apear:
`  Too many candidates found for 'dallas_id' type 'dallas::DallasComponent' You must assign an explicit ID to the parent component you want to use.`

This is definetly not the way , with i would like to see :)

ssieb, 2023-02-26:
If you have more than one dallas hub, you have to tell the sensor which one to use as explained in the docs.  How else would it work?

nagyrobi, 2023-02-27:
How about reading the [docs](https://esphome.io/components/sensor/dallas.html#dallas-component)?

Try accordingly:
```
dallas:
- id: "dallas_hub_1"
  pin: GPIO2
- id: "dallas_hub_2"
  pin: GPIO4

sensor:
- platform: dallas
  address: 0xA90...
  name: "Room1"
  dallas_id: dallas_hub_1
- platform: dallas
  address: 0xA1...
  name: "Room2"
  dallas_id: dallas_hub_2
```

MiG-41, 2023-02-27:
Thank you very much @nagyrobi ! 
Unfortunatly docs become too dificult to understand by me , but your example with id gives me ide very quick , and (with is most important ) works !

closed 2023-02-27 by MiG-41