esphome/feature-requests #1066: Trigger on loss of specific API connection

poldim, 2021-01-02:
**Describe the problem you have/What new integration you would like**
Identify that an ESPHome node has lost API connection to a specific HA instance.  When this happens, be able to trigger a set of actions such as `delay`, then check if still disconnnected, and turn off some switches.

**Please describe your use case for this integration and alternatives you've tried:**
I have a set of relays that control my heater.  If the heat is on, and the device loses WiFi/API connection, I want to be able to know that it lost as it will no longer receive the off state and continue to run the heater indefinitely.

**Additional context**
Contrary to https://github.com/esphome/issues/issues/394, it should only trigger based on this specific API connection.
I have a dev HA instance on my network so it would be good the be able to identify which API connection it has lost as the device does know which instance it is connected to:
`[D][api.connection:604]: Client 'Home Assistant 2020.12.2 (10.30.30.100)' connected successfully!`

0x3333, 2021-05-13:
Exactly the same issue here.

poldim, 2022-07-11:
> Duplicate of #1656

@nagyrobi  - It's close but it's not a duplicate.  #1656 does not appear to be asking to specify an HA instance by IP as one of many instances like this request is asking.  IE: I have a production and a development HA instances.  I only want the ESPHome device to care about the production HA, and not trigger any automations based on the dev instance.