esphome/feature-requests #1067: Add support for receiving the battery level for LYWSD02

xanderdekok, 2021-01-02:
**Describe the problem you have/What new integration you would like**

The LYWSD02 has support for receiving the Temperature and the Humidity, but reading the battery level is missing. Most other Xiaomi devices have the support for receiving the battery level.

The LYWSD02 does broadcast the battery level. I can see it in the logging of the ESP32:
[16:48:22][D][xiaomi_ble:300]: Got Xiaomi LYWSD02 (E7:2E:00:D1:A4:85):
[16:48:22][D][xiaomi_ble:309]:   Battery Level: 29%

Please add the support for receiving the battery level.

Thanks!

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

nikito7, 2021-03-06:
Use last dev

ChefkeGremmen, 2021-03-12:
I guess this ticket can be closed right? It is already working in version 1.16.0 in my setup.

xanderdekok, 2021-04-05:
It works for me also.

closed 2021-04-05 by xanderdekok