esphome/feature-requests #1072: Implement the ability to send data to thingspeak

gigios, 2021-01-04:
**Describe the problem you have/What new integration you would like**
Implement the ability to send data to thingspeak

**Please describe your use case for this integration and alternatives you've tried:**
I'm migrating a my arduino sketch to esphome. I'm able to port all my sensors,  even an MQ-135 gas sensor which is not available 'natively'.
The problem is that I have also the send all the data (fields) to a thingspeak channel. This is not possible now in ESPHome.
I have checked also Home Assistant, but it's possible to send only one field to a channel and not multiple fields.
My idea is to use ESPHome to implement my sensors and use Home Assistant as a supervisor, but I want that my sensors is able to works, and store data, also without the Home Assistant.
Do you think that this is possible or is it a wrong usage of ESPHome?

**Additional context**
Do you think that, eventually, it might be possible to use a 'fake' sensor ('[Generic Custom Component](https://esphome.io/custom/custom_component.html)'), to try to manage the sending of thingspeak data?

jbrepogmailcom, 2021-01-28:
Hello sir, I am not experienced with ESPHome, but I think it has HTTP GET component, so you can use that to post values to thingspeak, right? I use thingspeak a lot. Jan

gigios, 2021-01-30:
Thank you for your suggestion. Also i am not an expert, this is the first time i use ESPHome.
I will check the HTTP component.

frspp, 2021-04-07:
You can also use MQTT to send values to Thingspeak
https://esphome.io/components/mqtt.html
https://blogs.mathworks.com/iot/2017/01/20/use-mqtt-to-send-iot-data-to-thingspeak/
https://se.mathworks.com/help/thingspeak/mqtt-api.html

gigios, 2021-04-12:
Thanks for the answer, this for me it's a great news.
Thanks again, I will try immediately.

frspp, 2021-04-13:
@gigios please let me know if you get Thingspeak working.

I have tried setting mqtt publish to Thingspeak, no luck so far.

https://se.mathworks.com/help/thingspeak/mqtt-basics.html reads "MQTT publish does not require the API key." so i tried
`
mqtt:
  broker: mqtt.thingspeak.com

interval:
  - interval: 15sec
    then:
    - mqtt.publish:
        topic: channels/CHANNELID/publish/CHANNELWRITEAPIKEY 
        payload: field1=id
`
result: "MQTT Disconnected: Not Authorized."

I also looked https://se.mathworks.com/help/thingspeak/use-arduino-client-to-publish-to-a-channel.html for help but no luck. In this example they username (random), password (user write api key) and client id (random). 
`
mqtt:
  broker: mqtt.thingspeak.com
  username: usernamedoesnotmatter
  password: USERWRITEAPIKEY
  client_id: 202104132256
  discovery: false

interval:
  - interval: 15sec
    then:
    - mqtt.publish:
        topic: channels/CHANNELID/publish/CHANNELWRITEAPIKEY
        payload: field1=id
`
result: "MQTT Disconnected: Not Authorized."

jbrepogmailcom, 2021-04-13:
Hello I think it should be pretty easy to use current functionality and call thingspeak with http get. Have you tried it?

I personally would define global string variable and then update it in lambda with complete URL and field parameters. Then you can use http get in esphome to call thingspeak.

jbrepogmailcom, 2021-04-13:
You definitely need authorization to mqtt publish. From their web:
"MQTT Publish
The figure describes the topic structure. The Write API Key is required to publish. The broker acknowledges a correct CONNECT request with CONNACK"

ESPmonster, 2021-04-13:
@frspp  I have previously figured out the ESPhome -> thingspeak via MQTT problem. Have a look at my working code [here](https://github.com/ESPmonster/Esphome-boat-battery-and-waterlevel-monitor/blob/main/ESPhome-YAML)

It's essential to add 
`discovery_retain: 'false'`
and
`   discovery: False
    retain: false`

jbrepogmailcom, 2021-04-13:
Is it possible to connect to two mqtt brokers? If not then you can use thingspeak but you cannot send mqtt message to Home Assistant and have to use api. Or?

zierbeek, 2022-05-02:
Hi @ESPmonster , I'm also trying to get it published! but I have some trouble of selecting what I want to get published.

```
mqtt:
  broker: mqtt.thingspeak.com
  username: !
  password: !
  client_id: !
  discovery: false
  discovery_retain: 'false'
  discovery: False retain: false
  
interval:
  - interval: 3 sec
    then:
    - mqtt.publish:
        topic: channels/set/publish/set
        payload: field1="pH measurement"

sensor:
  - platform: ads1115
    multiplexer: 'A0_GND'
    gain: 6.144
    name: "pH measurement"
    unit_of_measurement: "pH"
    accuracy_decimals: 1
    state_class: "measurement"
    update_interval: 1s
    filters:
      - calibrate_linear:
        - 0.00 -> 0.0
        - 2.363 -> 7.0
        - 4.92 -> 14.0
```

zierbeek, 2022-05-02:
did someone get this working using http get?

ESPmonster, 2022-05-02:
@jbrepogmailcom unfortunately (AFAIK) you can only connect to one mqtt broker. But you can use mqtt + api  like you suggested.

@zierbeek See my code [here](https://github.com/ESPmonster/Esphome-boat-battery-and-waterlevel-monitor/blob/main/ESPHOME-YAML%20NEW). In this version I also implemented the possibility of subscribing the esphome device to the server and performing actions when receiving specific payloads (in my case, switching on a pump). It is important to realise that thingspeak will automatically disconnect as soon it received mqtt data with unknown topics. So, if you have frequent disconnects sniff the mqtt traffic using e.g. MQTT explorer and see if there are any useless messages. If this concerns sensors or switches try making them 'internal' or assign a proper topic that thingspeak understands.

ESPmonster, 2022-05-02:
This will probably work (just add your channel number where indicated):

> 
> mqtt:
>   broker: mqtt.thingspeak.com
>   username: !
>   password: !
>   client_id: !
>   discovery: false
>   discovery_retain: 'false'
>   discovery: False retain: false
>   
> 
> sensor:
>   - platform: ads1115
>     multiplexer: 'A0_GND'
>     gain: 6.144
>     name: "pH measurement"
>     state_topic: 'channels/<<INSERT CHANNEL #/publish/fields/field1'
>     unit_of_measurement: "pH"
>     accuracy_decimals: 1
>     state_class: "measurement"
>     update_interval: 3s
>     filters:
>       - calibrate_linear:
>         - 0.00 -> 0.0
>         - 2.363 -> 7.0
>         - 4.92 -> 14.0

zierbeek, 2022-05-02:
Thanks, unfortunately what I get is connection, and disconnection immediately.
TCP connection closed

ESPmonster, 2022-05-02:
No other sensors or switches etc. in your yaml? esphome will automatically send their measurements and status at nonsense topics to thingspeak which will trigger a disconnect. 

Also add the MQTT port. And remove the redundant "discovery: False retain: false" line. I suggest to use the new thingspeak server as the old is or will be deprecated (switch to mqtt3.thingspeak.com)

Like so: 

mqtt:
  broker: 'mqtt3.thingspeak.com'# alt: mqtt.thingspeak.com
  port: '1883'
  client_id: '<<INSERT CLIENT ID'#
  username: '<<INSERT USERNAME = CLIENT ID>>' # 
  password: '<<INSERT MQTT PASSWORD FROM THINGSPEAK>>'
  discovery: 'false'
  discovery_retain: 'false'
  birth_message: # leave empty
  will_message: # leave empty
  log_topic: #leave empty
  topic_prefix: 'channels'  # not sure whether this is necessary but works for me

zierbeek, 2022-05-02:
Well, yeah, there are multiple sensors for the ads to work. Thank you for your help!

`esphome:
  name: pool
  platform: ESP32
  board: nodemcu-32s

mqtt:
  broker: 'mqtt3.thingspeak.com'
  port: '1883'
  username: 'x'
  password: 'x'
  client_id: 'x'
  discovery: 'false'
  discovery_retain: 'false'

i2c:
  sda: 22
  scl: 21
  scan: true
  id: bus_a

ads1115:
  - address: 0x48
sensor:
  - platform: ads1115
    multiplexer: 'A0_GND'
    gain: 6.144
    name: "Po A0"
    unit_of_measurement: "V"
    state_class: "measurement"
    update_interval: 1s
  - platform: ads1115
    multiplexer: 'A0_GND'
    gain: 6.144
    state_topic: 'channels/1462749/publish/fields/field1'
    name: "pH measurement"
    unit_of_measurement: "pH"
    accuracy_decimals: 1
    state_class: "measurement"
    update_interval: 1s
    retain: false
    filters:
      - calibrate_linear:
        - 0.00 -> 0.0
        - 2.363 -> 7.0
        - 5.00 -> 14.0
  - platform: ads1115
    multiplexer: 'A1_GND'
    gain: 4.096
    id: do_measurement
    name: "Do A1"
    state_class: "measurement"
    unit_of_measurement: "V"
    update_interval: 1s
  - platform: ads1115
    multiplexer: 'A2_GND'
    gain: 6.144
    name: "To A2"
    state_class: "measurement"
    update_interval: 1s
  - platform: ads1115
    multiplexer: 'A2_GND'
    gain: 6.144
    name: "temperature"
    unit_of_measurement: "°C"
    state_class: "measurement"
    filters:
      - multiply: 7.44
    update_interval: 1s

binary_sensor:
  - platform: analog_threshold
    name: "Above set pH"
    sensor_id: do_measurement
    threshold: 2.41

web_server:
  port: 80

# Enable logging
logger:

# Enable Home Assistant API
# api:
#   password: ""

ota:
  password: ""

wifi:
  networks:
    ssid: Borremans 
    password: b0rr3m4ns

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Pool Fallback Hotspot"
    password: "1plusSwvK0"

captive_portal:
`

zierbeek, 2022-05-02:
I just keep on getting TCP disconnected, nothing pops up in Wireshark. Can't get Mqtt explorer to connect neither...

zierbeek, 2022-05-02:
[image]
This did successfully post to the thing speak