esphome/feature-requests #1073: Add a description or note field for each configuration on the dashboard

ttaggart4, 2021-01-05:
**Describe the problem you have/What new integration you would like**

Configuration names can be cryptic. A description would be helpful.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

glmnet, 2021-01-13:
https://esphome.io/components/esphome.html#configuration-variables

see the `comment:` option, is for dashboard .

If this is not what you're looking for, be more specific

closed 2021-01-13 by glmnet

ttaggart4, 2021-01-13:
The comment is only visible when you edit the node. I would like to see something on the main dashboard - a description (could be the same as comment) under or next to the node name. It currently shows the node name and file name, which are the same by default. Adding a description would allow for reminding the user where the node is physically in the house at a glance, and easily changed if the device is moved. I'm just learning ESPHome, there is a lot I don't know about it yet! Thanks :)

glmnet, 2021-01-13:
Ibb be guess you need to compile for the comment to show in the ui

ttaggart4, 2021-01-13:
Yes that works.