esphome/feature-requests #1074: 1-wire IButton reader integration

strbackatomas, 2021-01-06:
**Describe the problem you have/What new integration you would like**

The esphome device will report the iButton address attached to 1wire connected reader as value to home assitant sensor.

**Please describe your use case for this integration and alternatives you've tried:**

if I attach the iButton chip to the reader, the address of the chip will be displayed in the text sensor of the home assistant.
It can also be used as a trigger

**Additional context**

Possible sources:
https://github.com/pezinek/ibutton_reader
https://home.1vp.ru/en/1-wire-controller/
https://shop.loxone.com/cscz/ctecka-elektronickeho-klice.html

Misiu, 2021-01-07:
@strbackatomas great idea. I have a reader based on Arduino, but I'd like to move my project and integrate everything with Home Assistant.

Misiu, 2021-01-19:
This should be build based on https://github.com/esphome/esphome/tree/dev/esphome/components/dallas or should be added to this component, but not sure about the second option because 1-wire doesn't have a resolution and 1-wire isn't a temperature sensor.
@strbackatomas do you think the sensor should return value only when 1-wire ID is on the list of allowed sensors (like here: https://esphome.io/components/sensor/dallas.html) or all ID's and Home Assistant should decide if a specific value is allowed or not (like this: https://esphome.io/components/binary_sensor/pn532.html)?

So the config could look like this:

```
# Example configuration entry
onewire:
  - pin: 23

#individual sensors
sensor:
  - platform: onewire
    address: 0x1c0000031edd2a30
    name: "Dad keychain"
  - platform: onewire
    address: 0x2c0004431edd2a10
    name: "Mom keychain"
```

If you only add the first part then all IDs should get reported, if you add the second part then individual sensors should get created (as binary sensors?)
Not sure if this is possible, but that would allow the most flexible approach.

strbackatomas, 2021-01-19:
@Misiu both variants have their pros and cons. But a variant like PN532 would probably suffice.

fjufju, 2021-02-11:
Any updates on this.
It would be super usefull.
Home Assistant support tags, so I think 1-wire iButton could be used as a tag.

IvanGavrilenko, 2021-04-08:
May be as a temporary workaround anyone can add a custom sensor based on OneWire.h library?
As an example - http://robocraft.ru/blog/arduino/118.html
Sketch uploaded to wemos d1 shows on D3 pin and GND pin correct iButton address?

IvanGavrilenko, 2021-04-09:
As a temporary workaround:
create file onewire_sensor.h  near your yaml containing

```
#include "esphome.h" 
#include "OneWire.h"
#include <sstream>
#include <iomanip>

OneWire  ds(0); //0 - is GPIO number (on WEMOS D1 mini works fine on D3-GND without pull-up resistor - else needs 4k7 pull-up)

template<typename TInputIter>
std::string make_hex_string(TInputIter first, TInputIter last, bool use_uppercase = true, bool insert_semicolumn = true, bool insert_spaces = false)
{
    std::ostringstream ss;
    ss << std::hex << std::setfill('0');
    if (use_uppercase)
        ss << std::uppercase;
    while (first != last)
    {
        ss << std::setw(2) << static_cast<int>(*first++);
        if (insert_semicolumn && first != last)
            ss << ":";
        if (insert_spaces && first != last)
            ss << " ";
    }
    return ss.str();
}

class OneWireSensor : public PollingComponent, public TextSensor {
 public:
  // constructor
  OneWireSensor() : PollingComponent(150) {}

  void setup() override {
    // This will be called by App.setup()
  }
  void update() override {
    // This will be called every "update_interval" milliseconds.
    // Publish state

    byte addr[8];
    byte i;

	if ( !ds.search(addr)) {
      ds.reset_search();
      return;
	}

	if ( OneWire::crc8( addr, 7) != addr[7]) {
	  publish_state("CRC is not valid!\n");
	  return;
	}
  
	if ( addr[0] != 0x01) {
	  publish_state("Device is not a DS1990A family device.\n");
	  return;
	}
  
	publish_state(make_hex_string(std::begin(addr), std::end(addr)));
	delay(1000);
	publish_state("");

	ds.reset();

  }  
};
```

in yaml add
```
esphome:
  libraries:
   - "OneWire"
  includes:
    - onewire_sensor.h

text_sensor:
    - platform: custom
      lambda: |-
        auto my_custom_sensor = new OneWireSensor();
        App.register_component(my_custom_sensor);
        return {my_custom_sensor};
      text_sensors:
        name: "iButton" 

```

that would create custom text sensor that for 1 second would hold iButton adress in "01:02:03:04:05:06:AA:BA" format.
if the deletion of adress after 1 second is not needed - simply delete 2 lines from onewire_sensor.h:
```
	delay(1000);
	publish_state("");
```

fjufju, 2021-09-17:
Hi Team, any chance to have this? Home Assistant support tag, so iButton can work if this added to ESPHome.
Maybe next release 🤞

IvanGavrilenko, 2021-12-11:
Any news about adding iButton component?

Misiu, 2022-07-01:
@nagyrobi can we keep this open?

nagyrobi, 2022-07-01:
@IvanGavrilenko offered an alternative working solution for this request, I think it can be closed, however, discussion can continue.
Feel free to submit a PR to ESPHome with this as a component now.