esphome/feature-requests #1076: Support for ILI9342C display (M5 stack Core 2 display)

sj3fk3, 2021-01-07:
**Describe the problem you have/What new integration you would like**
The dev branch currently does support ILI9341 (but) the new M5Stack Core 2 device uses the slightly different ILI9342C display drivers. I would very much like to use the M5stack Core2 with ESPhome.

From what I have read, it is really similar to the ILI9341 with some some small changes.

http://www.buydisplay.com/download/ic/ILI9342.pdf

**Please describe your use case for this integration and alternatives you've tried:**

I cannot get the M5stack Core 2 display to work with the ILI9341 settings, it might need some slight changes to also support ILI9342C

**Additional context**
This discussion could help: https://www.eevblog.com/forum/microcontrollers/2-4-tft-screen-ili9342-problems-help-me-please-!/

kelchm, 2021-02-04:
I would love to see this as well. I don't have time at the moment, but may take a look at getting this implemented at some point in the next few weeks.

sj3fk3, 2021-02-15:
> I would love to see this as well. I don't have time at the moment, but may take a look at getting this implemented at some point in the next few weeks.

Do you have a Core 2? I'd be willing to ship you one if needed

dummylabs, 2021-03-12:
Yes, the support of ILI9342C would be highly desirable maybe even without a touch screen as a first step.

richardklingler, 2021-05-17:
Planned as one of the next components as I have some m5core2 here....

Not on the top list...but it also requires AXP PMIOC support as some of the control lines for the display run over this PMIC chip...

scargill, 2021-10-03:
Would be wonderful to have Core2 support as its a very neat boxed product ready to go...

martydingo, 2021-12-07:
+1 for support for AXP 192, the various libraries (custom_components) I have searched and tested across github & gitlab all result in odd and unexpected behaviour. 

Without this, half the functionality of the M5Stack Core2, M5Stick-C (and the respective PLUS model), and perhaps some other M5Stack boards, is completely missing, which makes the *entire* platform unusable. 

The M5Stack line of products seem to be making microelectronics more accessible to the general public, it really does make a lot of sense to put more priority into this to me.

martydingo, 2021-12-07:
+1 for support for AXP 192, the various libraries (custom_components) I have searched and tested across github & gitlab all result in odd and unexpected behaviour.

Without this, half the functionality of the M5Stack Core2, M5Stick-C (and the respective PLUS model), and perhaps some other M5Stack boards, is completely missing, which makes the entire platform unusable.

The M5Stack line of products seem to be making microelectronics more accessible to the general public, it really does make a lot of sense to add support

sj3fk3, 2021-12-08:
Could not agree more. It would be very nice when Nabu Casa and M5Core would do a joint venture. IMHO it's the missing link.

granvillebarker, 2021-12-17:
I think the main issue is the reset for the display on the Core2 is done on a GPIO line of the axp192 instead of the esp32, so there needs to be some way for the display functions to reset the display via the axp192.   I've been able to take some axp192 m5 stick code and alter it to power up the display, power the motor on and off, as well as send resets to the display, but I am not sure how to get the existing display library to use the axp192 code to reset the display.   In looking at the aurdino library that works with the core2 display, they appear to be using ili9341 library with this display other than the reset method.

trunneml, 2022-01-08:
+1 for M5Stack Core 2 support

granvillebarker, 2022-01-08:
I've recently been able to get the core2 to work using the standard ili9341 driver and a custome axp192 component, but it only works from power up.   Any reboot display is lost.

trunneml, 2022-01-08:
> I've recently been able to get the core2 to work using the standard ili9341 driver and a custom axp192 component, but it only works from power up.   Any reboot display is lost.  

Could you share your code? I'm currently trying the same. 😀

chrismetcalfe, 2022-01-08:
There's a custom axp192 component on gitlab that works with the M5 Stick C Plus. Seems to work well.

https://gitlab.com/geiseri/esphome_extras/-/tree/main/components/axp192

trunneml, 2022-01-09:
@granvillebarker I tried the turn on the display but failed. Could you share your axp192 component?

tuupola, 2022-01-10:
> I think the main issue is the reset for the display on the Core2 is done on a GPIO line of the axp192 instead of the esp32, so there needs to be some way for the display functions to reset the display via the axp192.

This is an annoyance with Core2. However the hard reset does not necessarily need be in the display functions. You can do it while [initialising the board](https://github.com/tuupola/esp_effects/blob/master/main/main.c#L257-L284) and then later do only [soft reset](https://github.com/tuupola/hagl_esp_mipi/blob/master/src/mipi_display.c#L226-L227) in the display functions. This is  not optimal but atleast there is no need to add axp192 dependency to the display functions.

martydingo, 2022-01-26:
> This is an annoyance with Core2. However the hard reset does not necessarily need be in the display functions. You can do it while [initialising the [...]

I do believe the obstacle is the path, and the obstacle is adding in support for the AXP192 modules, so it can interface with other modules. We shouldn't be trying to add in code for the AXP192 across loads of modules, rather modules should be able to invoke the AXP192 module to achieve their respective ends.

granvillebarker, 2022-01-26:
> I do believe the obstacle is the path, and the obstacle is adding in support for the AXP192 modules, so it can interface with other modules. We shouldn't be trying to add in code for the AXP192 [...]

I've seen where an axp192 module is in the works by one of the devs, but I don't think that has anything to do with the display issues.   The ili9341 driver works for me, if you power the module off and back on, but not if the soft reboots or the reset button is pressed, example on my github.   Other projects like openhasp work, so it's gotta be something simple.

martydingo, 2022-01-26:
> I've seen where an axp192 module is in the works by one of the devs, but I don't think that has anything to do with the display issues. The ili9341 driver works for me, if you power the module off [...]

AXP192 is a power management device. You are seeing issues with power management. Interfacing with the display may work using the ILI9341 driver, as after all, it is the driver. However, if the display is not in a state where the AXP192 needs to be communicated with to then interact with the ILI9341, then it won't work, the AXP192 will need initializing to then communicate with the ILI9341, as far as I understand. 

This has been the case on every M5 device I've used with ESPHome with an AXP192.

kpurintun, 2022-04-10:
ESP is awesome right up until you want to put one in the Wife's living areas and she think's its dumb looking.  these don't look bad at all.  especially the stack.  definitely excited for movement on this.

codedmind, 2022-04-25:
I cannot help... but found this https://github.com/m5stack/M5Unified
Maybe can push the integration into esphome?

Thanks

ultimateguy, 2022-05-07:
+1 here... bought this M5stack development kit not realizing it doesn't work with esphome :(

granvillebarker, 2022-05-07:
Check out [Openhasp](https://openhasp.haswitchplate.com/), it's one way to integrate the M5Core2 with Home Assistant until support is added.

codedmind, 2022-05-07:
I can't help integrate it into Espeon e, but if someone can the development version already have an auto configuration option that works

RamEliC, 2022-06-04:
Still trying to get this to work without much luck.  Please add support for M5 Stack core 2!!

nagyrobi, 2022-07-01:
https://github.com/esphome/feature-requests/issues/1721

KTibow, 2022-07-10:
Does anyone have this partially working?

rasclatt-dot-com, 2022-08-06:
+1 from me.
As per the feedback this is a bigger issue, a lot of the M5 devices include a high density of really useful sensors and features and are well packaged, however most of these features can only be enabled via the axp192 power managment chip, (screen, touch, microphone, amplifier, speaker, etc.
Also some of these are only accessiable via a p192 GPIO pins which can’t be referenced without the axp192 component.

I have tried a number of custom components for my M5core2 and have managed to get some components to work, however there are always unexpected results that make the solution unworkable, (the solution in the feed above allows speakers and less to work but switches on a constant vibration module).

There is real potential to use the M5Core2 as a couscous assistant (i2s microphone, built in speaker media players for genie/ acorn - combined with screen, sensors and built in leds) but until the power management is fixed these can’t be accessed)

martydingo, 2022-09-03:
> +1 from me. As per the feedback this is a bigger issue, a lot of the M5 devices include a high density of really useful sensors and features and are well packaged, however most of these features can [...]

I have gotten this to work with the display now, without that horrendous vibration. 

In `axp192.cpp`, on line 10, there is the following:

```c
void AXP192Component::setup() 
{
  begin(false, false, false, false, false);
}
```

Each false corresponds to a disableLDO variable. The vibration motor is on LDO2 on the Core2, so you can disable the motor by updating that line to the following:

```c
void AXP192Component::setup() 
{
  begin(false, true, false, false, false);
}
```

My configuration is as follows:

```yaml
esphome:
  name: m5stack-core2
  platform: ESP32
  board: m5stack-core-esp32

  includes:
   - /config/custom_components/axp192/axp192.h

wifi:
  ssid: "xxx"
  password: "xxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "M5Stack-Core2 Fallback Hotspot"
    password: "x"

substitutions:
  devicename: m5core2
  upper_devicename: M5Core2 

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

web_server:

# AXP192 power management - must be present to initialize TFT power on
sensor:
  - platform: axp192
    address: 0x34
    i2c_id: bus_a
    update_interval: 30s
    battery_level:
      name: "M5Core2 Battery Level"
      id: "m5core2_batterylevel"

  - platform: wifi_signal
    name: ${upper_devicename} WiFi Signal
    id: wifi_dbm
  - platform: uptime
    name: ${upper_devicename} Uptime

spi:
  clk_pin: GPIO18
  mosi_pin: GPIO23
  miso_pin: GPIO38
  
i2c:
   - id: bus_a
     sda: GPIO21
     scl: GPIO22
     scan: True

font:
  - file: 'fonts/arial.ttf'
    id: font1
    size: 8

# # builtin 80x160 TFT
display:
  - platform: ili9341
    model: M5STACK
    cs_pin: 5
    dc_pin: 15
    lambda: |-
      it.print(80, 0, id(font1), TextAlign::TOP_CENTER, "M5Stick Test");
```

And it works!

cmet7, 2022-09-03:
> And it works!

Just tried this and it's still not working after reboot.

martydingo, 2022-09-03:
Aha, I've ran into an issue after a reboot, to do with the 

> > And it works!
> 
> Just tried this and it's still not working after reboot.

I've just ran into the issue after posting, following modifications to the original library to add model support.

I've fixed the issue, it should compile however the issue is to do with the backlight pin not getting any voltage, since the M5Sick and the Core2 have different pins for their backlights. 

I'm about to commit the code and I'll update soon.

martydingo, 2022-09-03:
Please let me know if there any issues with the code. I am seeing issues with the LCD on the M5Stick-C, but the Core2 works perfectly with the configuration within sample-config. 

Note to update the wifi & AP details as you see fit :) 

https://github.com/martydingo/esphome-axp192

cmet7, 2022-09-08:
> the Core2 works perfectly with the configuration within sample-config.

The sample config for core 2 is empty.

martydingo, 2022-09-09:
> > the Core2 works perfectly with the configuration within sample-config.
> 
> The sample config for core 2 is empty.

Okay, that's embarrassing, fixed now.

clowrey, 2022-09-28:
Now we just need touchscreen driver for the FT6336U IC :) I don't have a core 2 but the WT32-SC01 which I have also uses this touch IC..

spali, 2022-09-28:
BTW for others:
The [M5GO IoT Starter Kit v2.6](https://shop.m5stack.com/products/m5go-iot-starter-kit-v2-6) has a ILI9342C and Pwr Mgmt IP5306.
It works with following settings:
```yaml
external_components:
  - source:
      type: git
      url: https://github.com/ssieb/custom_components
    components: [ip5306]
ip5306:
  battery_level: # sensor
    name: ${upper_devicename} Battery Level
    entity_category: diagnostic
  charger_connected: # binary_sensor
    name: ${upper_devicename} Charger Connected
    entity_category: diagnostic
  charge_full: # binary_sensor
    name: ${upper_devicename} fully charged
    entity_category: diagnostic
display:
  - platform: ili9341
    model: M5STACK
    #model: TFT_2.4R (mirrors and inverts the display)
    cs_pin: GPIO14
    dc_pin: GPIO27
    reset_pin: GPIO33
    led_pin: GPIO32
    rotation: 0
```
I don't know the technical details behind display, and power mgmt and if they are related (which is different regarding the normal Core2). But it doesn't work with commented out TFT_2.4R setting.

cmet7, 2022-10-09:
> > > the Core2 works perfectly with the configuration within sample-config.
> > 
> > 
> > The sample config for core 2 is empty.
> 
> Okay, that's embarrassing, fixed now.

@martydingo I haven't been able to get this working. I am seeing this error: `Platform not found: 'sensor.axp192'.`

I'm using your sample config for the Core 2.

cmet7, 2022-10-10:
For what it's worth I got this working by cloning the git repo referenced above and moving the `axp192` directory under `/custom_components`.  I'm running ESP Home in a docker container and the include seems not to work.

I can confirm that this is working on M5 Stack Core2 and the display continues working even after a reset.

landonr, 2022-10-10:
you can include the axp192 component from gitlab like this. this works for m5stick-c
```
external_components:
  - source:
      type: git
      url: https://gitlab.com/geiseri/esphome_extras.git
    refresh: 0s
    components: [axp192]

axp192:
  id: axp
  address: 0x34
  i2c_id: bus_a
  update_interval: 60s

sensor:
    # AXP192 power management - must be present to initialize TFT power on
  - platform: axp192
    axp192_id: axp
    id: batteryPercent

light:
  - platform: axp192
    id: axpbacklight
    axp192_id: axp
    restore_mode: ALWAYS_ON
    name: Backlight

binary_sensor:
  - platform: axp192
    axp192_id: axp
    type: charging
    id: axp_charger
    name: Charger
```

cmet7, 2022-10-11:
@martydingo this is working well with the exception of _brightness_. Core 2 uses DC-DC3 for brightness control which doesn't work the same way that LDO2 works. DC-DC3 is controlled by the 7 least significant bits in the 0x27 register; the most significant bit is reserved. As a result the brightness value shifts left 3 places rather than 4. In order to set the register I read the entire byte, logical AND with 0x80 to clear all bits except the reserved bit, logical OR with the shifted brightness value.

I've opened a pull request against your repo.

dereisele, 2022-11-23:
I've got a M5Stack Core Gray with the same display. Here is my first **not working** try: https://github.com/dereisele/esphome_components

The Display shows a white bar on the right side

[image]

martydingo, 2022-11-29:
@cmet7 thanks very much for that, I've merged the PR after taking a gander.

Apologies for the delay on that!

martydingo, 2022-11-29:
> I've got a M5Stack Core Gray with the same display. Here is my first **not working** try: https://github.com/dereisele/esphome_components
> 
> The Display shows a white bar on the right side [image]

Please supply all code, not only the custom component but your ESPHome YAML too please. :)

dereisele, 2022-12-01:
@martydingo Here is my config

```yaml
esphome:
  name: m5stack

external_components:
#  - source:
#      type: git
#      url: https://github.com/dereisele/esphome_components
#    components: [ ili9341 ]
  - source:
      type: "local"
      path: "../../esphome_components/components"
    components:
      - ili9341

esp32:
  board: m5stack-grey
  framework:
    type: arduino

# Enable logging
logger:

mqtt:
  broker: !secret mqtt_host
  discovery: true
  username: !secret mqtt_user
  password: !secret mqtt_password

# Enable Home Assistant API
api:
  encryption:
    key: "xxxx"

ota:
  password: "xxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "M5Stack Fallback Hotspot"
    password: "xxx"

captive_portal:

spi:
  clk_pin: 18
  mosi_pin: 23
  miso_pin: 19

font:
  - file: "fonts/Roboto-Medium.ttf"
    id: font_roboto_medium22
    size: 22
    glyphs: '!"%()+,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz/³µ'

display:
  - platform: ili9xxx
    id: m5stack_display
    model: M5STACKGREY
    cs_pin: 14
    dc_pin: 27
    backlight_pin: 32
    led_pin: 33
    rotation: 0
    lambda: |-
      Color RED(1,0,0);
      Color BLUE(0,0,1);
      Color WHITE(1,1,1);
      it.rectangle(0,  0, it.get_width(), it.get_height(), BLUE);
      it.rectangle(0, 22, it.get_width(), it.get_height(), RED);

# Button to toggle the display backlight
binary_sensor:
  - platform: gpio
    id: M5_BtnA
    pin:
      number: 39
      inverted: true
  - platform: gpio
    id: M5_BtnB
    pin:
      number: 38
      inverted: true
  - platform: gpio
    id: M5_BtnC
    pin:
      number: 37
      inverted: true
    on_click:
      then:
        - switch.toggle: backlight

# GPIO pin of the display backlight
switch:
  - platform: gpio
    pin: 32
    name: "Backlight"
    id: backlight
    restore_mode: ALWAYS_ON
```

dereisele, 2022-12-01:
> could you try my changes to the ili9341 component at [esphome/esphome#3795](https://github.com/esphome/esphome/pull/3795)
> 
> this should fix it i belief.

With that config I get a complete white screen (ili9342 model config)

nielsnl68, 2022-12-01:
@dereisele do you mean you get ONLY the white screen or do you also get the colored rectangles?

dereisele, 2022-12-02:
> @dereisele do you mean you get ONLY the white screen or do you also get the colored rectangles?

No, just a complete white screen

leoedin, 2022-12-06:
I have an M5Core2 - and was seeing a similar white bar to @dereisele on startup. I had a look through the Arduino startup code for the M5Core2 and it seems that they toggle the reset pin (pulling it low) on the display for 100ms. The ESPHome ILI9341 driver also does this. On the M5Core2 the display reset pin is on GPIO4 of the PMC.

I've modified the component from @martydingo to do this reset, and the display appears to work reliably now.

The changes are here: https://github.com/leoedin/esphome-axp192/blob/master/components/axp192/axp192.cpp#L86 

And my config is:

`esphome:
  name: 'm5core'
  platform: ESP32
  board: m5stack-core-esp32

external_components:
  - source:
     type: local
     path: custom_components/esphome-axp192/components

  # # - source:
#      type: git
#      url: 'https://github.com/martydingo/esphome-axp192'
#    components: [axp192]

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

wifi:
  ssid: "yourwifi"
  password: "blabla"

# Define hardware
# I2C bus used to talk to AXP192
i2c:
  - id: bus_a
    sda: GPIO21
    scl: GPIO22

# AXP192 power management chip controls everything
sensor:
  - platform: axp192
    model: M5CORE2
    address: 0x34
    i2c_id: bus_a
    update_interval: 30s
    brightness: 75%
    battery_level:
      name: "Device Battery Level"
      id: "device_batterylevel"

# SPI talks to display
spi:
  clk_pin: GPIO18
  mosi_pin: GPIO23
  miso_pin: GPIO38

# ILI9342C
display:
  - platform: ili9341
    model: M5STACK
    cs_pin: GPIO5
    dc_pin: GPIO15
    lambda: |-
      it.print(80, 0, id(font_roboto_20), TextAlign::TOP_CENTER, "M5Core Test");

# Fonts (required for display)
font:
  - file: "gfonts://Roboto"
    id: font_roboto_20
    size: 20`

dereisele, 2022-12-06:
@leoedin I got my M5Stack Gray working with your changes! Thank you very much!!!

```yaml
esphome:
  name: m5stack

external_components:
  - source:
      type: git
      url: 'https://github.com/leoedin/esphome-axp192'
    components: [axp192]

i2c:
  - id: bus_a
    sda: GPIO21
    scl: GPIO22

sensor:
  - platform: axp192
    model: M5CORE2
    address: 0x34
    i2c_id: bus_a
    update_interval: 30s
    brightness: 75%
    battery_level:
      name: "Device Battery Level"
      id: "device_batterylevel"

esp32:
  board: m5stack-grey
  framework:
    type: arduino

# Enable logging
logger:

mqtt:
  broker: !secret mqtt_host
  discovery: true
  username: !secret mqtt_user
  password: !secret mqtt_password

# Enable Home Assistant API
api:
  encryption:
    key: "xxx"

ota:
  password: "xxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "M5Stack Fallback Hotspot"
    password: "xxx"

captive_portal:

spi:
  clk_pin: 18
  mosi_pin: 23
  miso_pin: 19

font:
  - file: "gfonts://Roboto"
    id: font_roboto_20
    size: 20

# ILI9342C
display:
  - platform: ili9341
    model: M5STACK
    cs_pin: 14
    dc_pin: 27
    led_pin: 33
    lambda: |-
      it.print(80, 0, id(font_roboto_20), TextAlign::TOP_CENTER, "M5Core Test");

# Button to toggle the display backlight
binary_sensor:
  - platform: gpio
    id: M5_BtnA
    pin:
      number: 39
      inverted: true
  - platform: gpio
    id: M5_BtnB
    pin:
      number: 38
      inverted: true
  - platform: gpio
    id: M5_BtnC
    pin:
      number: 37
      inverted: true
    on_click:
      then:
        - switch.toggle: backlight

# GPIO pin of the display backlight
switch:
  - platform: gpio
    pin: 32
    name: "Backlight"
    id: backlight
    restore_mode: ALWAYS_ON
```

dereisele, 2022-12-06:
But my Battery Level randomly jumps to -247,1 %

leoedin, 2022-12-07:
I didn't validate battery level - so it's possible it's a bug. It could also be because the Gray probably has different pins for different things on the PMC. You'll have to look at the schematic from M5 and compare it with the Core2.

dereisele, 2022-12-07:
I just noticed, that the other configs always include this line and the axp192 sensor config 🤦‍♂️ This is probably why it works now

```
# AXP192 power management - must be present to initialize TFT power on
```

Archef2000, 2022-12-17:
> I have an M5Core2 - and was seeing a similar white bar to @dereisele on startup. I had a look through the Arduino startup code for the M5Core2 and it seems that they toggle the reset pin (pulling it [...]

I can't get it to work with my M5Stack Core 2 i used your config and cloned your linked repo but it just does noting

Archef2000, 2022-12-17:
So far i got the MPU, Touch, Battery(not accrurate) working but backlight and display is only flashing shortly after the beginning.
```
esphome:
  name: core2

esp32:
  board: m5stack-core2
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: ""

ota:
  password: ""

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Fallback Hotspot"
    password: !secret wifi_password

captive_portal:

web_server:
  port: 80
 
external_components:
  - source: github://gpambrozio/esphome@FT6336U-touch
    components: [ ft63x6 ]
  - source: github://TomG736/esphome-BM8563
    components: [ bm8563 ]
  - source:
     type: local
     path: custom_components/leoedin-esphome-axp192-master/components

#   - source:
#      type: git
#      url: 'https://github.com/leoedin/esphome-axp192'
#    components: [axp192]

touchscreen:
  - platform: ft63x6
    i2c_id: bus_a
    on_touch:
      - logger.log:
          format: Touch at (%d, %d)
          args: [touch.x, touch.y]

spi:
  clk_pin: GPIO18
  mosi_pin: GPIO23
  miso_pin: GPIO38

i2c:
  - id: bus_a
    sda: GPIO21
    scl: GPIO22

sensor:
  - platform: axp192
    model: M5CORE2
    address: 0x34
    i2c_id: bus_a
    update_interval: 3s
    brightness: 75%
    battery_level:
      name: "Device Battery Level"
      id: "device_batterylevel"
  
  - platform: mpu6886
    address: 0x68
    accel_x:
      name: "MPU6886 Accel X"
    accel_y:
      name: "MPU6886 Accel Y"
    accel_z:
      name: "MPU6886 Accel z"
    gyro_x:
      name: "MPU6886 Gyro X"
    gyro_y:
      name: "MPU6886 Gyro Y"
    gyro_z:
      name: "MPU6886 Gyro z"
    temperature:
      name: "MPU6886 Temperature"

binary_sensor:
  - platform: touchscreen
    name: Top Left Touch Button
    x_min: 0
    x_max: 100
    y_min: 0
    y_max: 100

  - platform: gpio
    id: touching
    name: Touching
    pin:
      number: 39
      inverted: true

## Not working
color:
  - id: my_red
    red: 100%
    green: 3%
    blue: 5%

font:
  - file: "gfonts://Roboto"
    id: my_font
    size: 20

binary_sensor:
  - platform: touchscreen
    name: Top Left Touch Button
    x_min: 0
    x_max: 100
    y_min: 0
    y_max: 100

  - platform: gpio
    id: touching
    name: Touching
    pin:
      number: 39
      inverted: true

display:
  - platform: ili9341
    model: M5STACK
    cs_pin: GPIO5
    dc_pin: GPIO15
    lambda: |-
      it.print(80, 0, id(my_font), id(my_red), TextAlign::TOP_CENTER, "Hello World!");
```

nielsnl68, 2022-12-18:
You could try to add:
```yaml
  led_pin: GPIO32
```
to your display component. That should enable your backlight.

leoedin, 2022-12-18:
I think we're talking about 2 different boards here. The M5 Stack Core 2 doesn't have the display LED driven by the ESP32 GPIO - it's driven by a pin on the PMC. It's probably best to be clear what board you're talking about.

@Archef2000 I wonder if it's a problem with the code for the PMC and the touch controller? Have you tried removing the ft63x6 component? Just looking at the ft63x6 component implementation, it doesn't seem to do any I2C bus setup or have the correct I2C address in it (0x38 according to the M5Core2 code on github). Perhaps I'm misunderstanding the implementation? I'm quite new to ESPHome components. 

If it's not doing proper setup of the I2C bus then it could be clashing with the PMC.

Archef2000, 2022-12-18:
So it is the Core 2 as mentioned in the comment above.

Archef2000, 2022-12-18:
When i flash the esp i get following error

```
WARNING GPIO15 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
```

nagyrobi, 2022-12-18:
It's not an error, it's a `WARNING`.

Archef2000, 2022-12-18:
The problem was the axp192 component from @leoedin i now use the component from @martydingo.
```yaml
esphome:
  name: core2

esp32:
  board: m5stack-core2
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "9p2da+QiKqWc9v6ZGoQ0VgPHEBmtvdZsUJ+rjbu4iSQ="

ota:
  password: "ab8ee613745c24ba1e16ac7bbef8afe0"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Core2 Fallback Hotspot"
    password: !secret wifi_password

captive_portal:

web_server:
  port: 80

external_components:
  - source: github://gpambrozio/esphome@FT6336U-touch
    components: [ ft63x6 ]
  - source: github://TomG736/esphome-BM8563
    components: [ bm8563 ]
  - source: github://martydingo/esphome-axp192
    components: [axp192]

touchscreen:
  - platform: ft63x6
    i2c_id: bus_a
    on_touch:
      - logger.log:
          format: Touch at (%d, %d)
          args: [touch.x, touch.y]

spi:
  clk_pin: GPIO18
  mosi_pin: GPIO23
  miso_pin: GPIO38

i2c:
  - id: bus_a
    sda: GPIO21
    scl: GPIO22

sensor:
  - platform: axp192
    model: M5CORE2
    address: 0x34
    i2c_id: bus_a
    update_interval: 3s
    brightness: 75%
    battery_level:
      name: "Device Battery Level"
      id: "device_batterylevel"
  
  - platform: mpu6886
    address: 0x68
    accel_x:
      name: "MPU6886 Accel X"
    accel_y:
      name: "MPU6886 Accel Y"
    accel_z:
      name: "MPU6886 Accel z"
    gyro_x:
      name: "MPU6886 Gyro X"
    gyro_y:
      name: "MPU6886 Gyro Y"
    gyro_z:
      name: "MPU6886 Gyro z"
    temperature:
      name: "MPU6886 Temperature"

color:
  - id: my_red
    red: 100%
    green: 3%
    blue: 5%

font:
  - file: "gfonts://Roboto"
    id: my_font
    size: 20

binary_sensor:
  - platform: touchscreen
    name: Left Touch Button
    x_min: 0
    x_max: 106
    y_min: 120
    y_max: 140
  - platform: touchscreen
    name: Middle Touch Button
    x_min: 107
    x_max: 213
    y_min: 120
    y_max: 140
  - platform: touchscreen
    name: Right Touch Button
    x_min: 214
    x_max: 320
    y_min: 120
    y_max: 140
  
  - platform: gpio
    id: touching
    name: Touching
    pin:
      number: 39
      inverted: true

display:
  - platform: ili9341
    model: M5STACK
    cs_pin: GPIO5
    dc_pin: GPIO15
    lambda: |-
      it.print(80, 0, id(my_font), id(my_red), TextAlign::TOP_CENTER, "Hello World!");

```
Now i only got two problems:
  -  How to createe a multipage ui
  -  Turn off the touch sensors after release

Archef2000, 2022-12-18:
The problem was the axp192 component from @leoedin i now use the component from @martydingo.
```yaml
esphome:
  name: core2

esp32:
  board: m5stack-core2
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: ""

ota:
  password: ""

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Fallback Hotspot"
    password: !secret wifi_password

captive_portal:

web_server:
  port: 80

external_components:
  - source: github://gpambrozio/esphome@FT6336U-touch
    components: [ ft63x6 ]
  - source: github://TomG736/esphome-BM8563
    components: [ bm8563 ]
  - source: github://martydingo/esphome-axp192
    components: [axp192]

touchscreen:
  - platform: ft63x6
    i2c_id: bus_a
    on_touch:
      - logger.log:
          format: Touch at (%d, %d)
          args: [touch.x, touch.y]

spi:
  clk_pin: GPIO18
  mosi_pin: GPIO23
  miso_pin: GPIO38

i2c:
  - id: bus_a
    sda: GPIO21
    scl: GPIO22

sensor:
  - platform: axp192
    model: M5CORE2
    address: 0x34
    i2c_id: bus_a
    update_interval: 3s
    brightness: 75%
    battery_level:
      name: "Device Battery Level"
      id: "device_batterylevel"
  
  - platform: mpu6886
    address: 0x68
    accel_x:
      name: "MPU6886 Accel X"
    accel_y:
      name: "MPU6886 Accel Y"
    accel_z:
      name: "MPU6886 Accel z"
    gyro_x:
      name: "MPU6886 Gyro X"
    gyro_y:
      name: "MPU6886 Gyro Y"
    gyro_z:
      name: "MPU6886 Gyro z"
    temperature:
      name: "MPU6886 Temperature"

color:
  - id: my_red
    red: 100%
    green: 3%
    blue: 5%

font:
  - file: "gfonts://Roboto"
    id: my_font
    size: 20

binary_sensor:
  - platform: touchscreen
    name: Left Touch Button
    x_min: 0
    x_max: 106
    y_min: 120
    y_max: 140
  - platform: touchscreen
    name: Middle Touch Button
    x_min: 107
    x_max: 213
    y_min: 120
    y_max: 140
  - platform: touchscreen
    name: Right Touch Button
    x_min: 214
    x_max: 320
    y_min: 120
    y_max: 140
  
  - platform: gpio
    id: touching
    name: Touching
    pin:
      number: 39
      inverted: true

display:
  - platform: ili9341
    model: M5STACK
    cs_pin: GPIO5
    dc_pin: GPIO15
    lambda: |-
      it.print(80, 0, id(my_font), id(my_red), TextAlign::TOP_CENTER, "Hello World!");

```
Now i only got two problems:
  -  How to createe a multipage ui
  -  Turn off the touch sensors after release

[53 more replies]