esphome/feature-requests #1077: [Light RGBCCT Support] Warm and cold white controlled by a CT and brightness channel.

jaddel, 2021-01-07:
**Describe the problem you have/What new integration you would like**

I got a bulb that can't be controlled with the RGBWW Light since the cold and warm white channels are split into a "color temperature" and a "brightness" channel. 

Bulb: Athom 15W [https://templates.blakadder.com/athom_LB01-15W-E27.html](url)

**Please describe your use case for this integration and alternatives you've tried:**

Tasmota provides an option to control these kind of bulbs but I'd rather like to stick with esphome since everything else is already based on esphome in my environment. 

**Additional context**
This is what i got so far to make bulb behave as expected using Home-assistant to control it. 

The code seems to work fine so far unless the color_interlock dosen't apply when using the RGB channels. 

ESPHome Yaml Config:

`substitutions:
  device_name: bulb_athom_04
  friendly_name: Wall
esphome:
  name: bulb_athom_04
  platform: ESP8266
  board: esp01_1m
  includes:
  - includes/athom_15w.h
  arduino_version: espressif8266@2.6.2
  build_path: bulb_athom_04
  platformio_options: {}
  esp8266_restore_from_flash: false
  board_flash_mode: dout
  libraries: []
wifi:
  networks:
  - ssid: !secret 'wifi_ssid'
    password: !secret 'wifi_password'
    priority: 0.0
  reboot_timeout: 24h
  fast_connect: true
  ap:
    ssid: bulb_athom_04
    password: !secret 'wifi_ap_password'
    ap_timeout: 1min
  domain: .local
  power_save_mode: NONE
  output_power: 20.0
  use_address: bulb_athom_04.local
captive_portal: {}
web_server:
  port: 80
  css_url: https://esphome.io/_static/webserver-v1.min.css
  js_url: https://esphome.io/_static/webserver-v1.min.js
logger:
  baud_rate: 0
  tx_buffer_size: 512
  hardware_uart: UART0
  level: DEBUG
  logs: {}
  esp8266_store_log_strings_in_flash: true
api:
  reboot_timeout: 24h
  port: 6053
  password: ''
ota:
  safe_mode: true
  port: 8266
  password: ''
output:
- platform: esp8266_pwm
  pin:
    number: 13
    mode: OUTPUT
    inverted: false
  id: out_cw
  frequency: 1000.0
- platform: esp8266_pwm
  pin:
    number: 5
    mode: OUTPUT
    inverted: false
  id: out_b
  frequency: 1000.0
- platform: esp8266_pwm
  pin:
    number: 4
    mode: OUTPUT
    inverted: false
  id: out_red
  frequency: 1000.0
- platform: esp8266_pwm
  pin:
    number: 12
    mode: OUTPUT
    inverted: false
  id: out_green
  frequency: 1000.0
- platform: esp8266_pwm
  pin:
    number: 14
    mode: OUTPUT
    inverted: false
  id: out_blue
  frequency: 1000.0
light:
- platform: custom
  lambda: !lambda "auto light_out = new Athom15w(id(out_cw),id(out_b),id(out_red),id(out_green),id(out_blue));\
    \ \nApp.register_component(light_out); \nreturn {light_out}; "
  lights:
  - name: Wall
    gamma_correct: 0.0
    restore_mode: RESTORE_DEFAULT_OFF
    default_transition_length: 1s`

ESP C++  - includes/athom_15w.h Config:

`#pragma once

#include "esphome.h"

class Athom15w : public Component, public LightOutput {
 public:
  Athom15w(FloatOutput *cw_white_color, FloatOutput *brightness, FloatOutput *red, FloatOutput *green, FloatOutput *blue)
    {
        cw_white_color_ = cw_white_color;
        brightness_ = brightness;
        red_ = red;
        green_ = green;
        blue_ = blue;
        cold_white_temperature_ = 175;
        warm_white_temperature_ = 333;
        constant_brightness_ = true;
        color_interlock_ = true;
    }
  LightTraits get_traits() override {
    auto traits = light::LightTraits();
    traits.set_supports_brightness(true);
    traits.set_supports_rgb(true); 
    traits.set_supports_rgb_white_value(true); 
    traits.set_supports_color_interlock(true); 
    traits.set_supports_color_temperature(true);
    traits.set_min_mireds(this->cold_white_temperature_);
    traits.set_max_mireds(this->warm_white_temperature_);
    return traits;
  }

  void write_state(LightState *state) override {
    float brightness, cwhite, wwhite, red, green, blue;

    state->current_values_as_rgbww(&red, &green, &blue, &cwhite, &wwhite, this->constant_brightness_, this->color_interlock_);

//    ESP_LOGD("custom", "The value cwhite: %f", cwhite);
//    ESP_LOGD("custom", "The value wwhite: %f", wwhite);
    state->current_values_as_brightness(&brightness);
//    ESP_LOGD("custom", "The value brightness: %f", brightness);
    this->cw_white_color_->set_level(cwhite+(1-brightness));
    this->brightness_->set_level(brightness);
    this->red_->set_level(red);
    this->green_->set_level(green);
    this->blue_->set_level(blue);
  }
  protected:
    FloatOutput *cw_white_color_;
    FloatOutput *brightness_;
    FloatOutput *red_;
    FloatOutput *green_;
    FloatOutput *blue_;
    float cold_white_temperature_;
    float warm_white_temperature_;
    bool constant_brightness_;
    bool color_interlock_;
};`

jaddel, 2021-01-28:
https://github.com/jaddel/ESPHome/tree/master/Devices/Bulb%20Athom%2015W%20RGBWCT

ranomier, 2021-04-10:
Why did you close this?
There should be a feature in esphome for this. I have these stupid things: https://templates.blakadder.com/lvwit_A70_1521lm.html

https://tasmota.github.io/docs/Lights/#pwm-ct
This is the option from tasmota that is missing in esphome.

I will try your library with my lights we'll see how it works.

jaddel, 2021-04-10:
> Why did you close this?
> There should be a feature in esphome for this. I have these stupid things: https://templates.blakadder.com/lvwit_A70_1521lm.html [...]

hey, yeah you are welcome to use it. 

Sorry for closing I'm not jet used the whole github process...

mew1033, 2021-04-12:
This is very similar to my feature request: in #1131. It'd be awesome to allow for full temperature control with just a single color white light + RGB.
See the Tasmota feature right next to the referenced above: https://tasmota.github.io/docs/Lights/#virtual-ct

WeekendWarrior1, 2021-09-27:
Isn't this now supported with:
https://esphome.io/components/light/rgbct.html?

jaddel, 2021-09-27:
> Isn't this now supported with:
> https://esphome.io/components/light/rgbct.html?

yes, I'll close the issue.

closed 2021-09-27 by jaddel