esphome/feature-requests #1078: Add support for Qingping CGDK2 Temp & RH Monitor Lite

DarkFeniXZp, 2021-01-09:
**Describe the problem you have/What new integration you would like**
I have this device similar to Qingping CGG1.  But it did not work with esphome through the integration for such sensor.
**Please describe your use case for this integration and alternatives you've tried:**
I found the only mention of working with this sensor in the [issue for ATC_MiThermometer](https://github.com/atc1441/ATC_MiThermometer/issues/135).  Some developments on this sensor are already mentioned there.

**Additional context**

cervenca, 2021-01-30:
I have several Qingping products if needed to generate BT logs. 
CGDK2 LITE version, ink round.
thank you.

TheGroundZero, 2021-02-04:
I recently bought 3 CGG1's expecting to be able to get their data into Home Assistant via ESPHome (flashing custom firmware if necessary). But now it appears they aren't compatible yet? :(

[image]
https://www.aliexpress.com/item/32994401600.html

DarkFeniXZp, 2021-03-16:
I found that CGDK2 support was introduced in OpenGateway.  Perhaps this will help add CGDK2 support to Esphome.
https://github.com/1technophile/OpenMQTTGateway/pull/884/commits/2c2de8970698ed82e2e8afd1bdfce969ada54146

cervenca, 2021-03-16:
I have found a way to connect to MI app, and get the bind key, as messages are encrypted.
http://www.d0wn.com/qingping-cleargrass-cgdk2-factory-reset/
also there is an workaround in case you get stuck.

and from here the app for android that worked on my test mobile:
https://apkvision.com/app/modified-mi-home-by-vevs-38331/

I try to study how this is working:
https://github.com/custom-components/ble_monitor
to decode the messages from 2 or 3 devices and then see how to modify esphome to accept also cgdk2 version.

Interesting to read code from esphome and identify where to modify.
depending on my available time will progress on this, else I am able to provide logs to anyone that can speed up the process.

cervenca, 2021-03-16:
it seems that there is the code here:
https://github.com/zaluthar/esphome/commit/efb6a012c59a7cca10dc7e0b6610baefa13a629d
thanks @zaluthar

cervenca, 2021-03-17:
I was able to use the code from zaluthar with some small modifications see comments:
https://github.com/zaluthar/esphome/commit/efb6a012c59a7cca10dc7e0b6610baefa13a629d
I was able to modify my code and test it. works like a charm now for 23h.

martinrosenauer, 2021-05-10:
@cervenca , can we somehow get your changes into the main branch ?

TheGroundZero, 2021-05-17:
> 
> 
> @cervenca , can we somehow get your changes into the main branch ?

Correct me if I'm wrong, but isn't CGG1 support added since 1.17.1 now?
Or is this only in the docs?

https://github.com/esphome/esphome-docs/pull/1136

martinrosenauer, 2021-05-17:
We are talking CGDK2 ;-)

TheGroundZero, 2021-05-17:
> 
> 
> We are talking CGDK2 ;-)

🤐 🤦‍♂️

hmax42, 2021-11-01:
paired it to the qingping app
after that i could use the webpage to get the bindkey

hmax42, 2021-11-03:
but the battery value looks weird

hmax42, 2021-11-03:
but the power value looks weird
[image]

hmax42, 2021-11-05:
still weird, but somehow expected
[image]

nagyrobi, 2022-07-19:
Implemented: https://esphome.io/components/sensor/xiaomi_ble.html?highlight=cgdk2#cgdk2

closed 2022-07-19 by nagyrobi