esphome/feature-requests #1081: Preserve/restore uptime across deep_sleep

ckuethe, 2021-01-10:
**Describe the problem you have/What new integration you would like**

An option to preserve uptime before engaging deep_sleep, and restoring the value on wake.

**Please describe your use case for this integration and alternatives you've tried:**

My use case is to track total time on battery for a battery operated sensor that spends most of its time sleeping... kind of like wall time vs. cpu time.

I'm new to esphome, so I'm not sure if I could do something like:
* create a global variable in rtc ram to store accumulated uptime
* `on_boot` restore the value of accumulated uptime from rtc ram
* `filter` the uptime to add the accumulated uptime before publishing the value to the dashboard
* `on_shutdown` increment the value of the accumulate uptime in rtc ram by the amount of the latest waking episode.

And then when I don't see updates from this sensor for some amount of time, I know it's time to go recharge the batteries.

**Additional context**

glmnet, 2021-01-13:
you should do that, yes. May be there are other options. If you need help with that I highly recommend you the discord chat, find the link in the GH readme page

closed 2021-01-13 by glmnet

dedalodaelus, 2021-06-05:
I leave here a an example with this done, if anyone find it useful:

[Esphome uptime preserve during deepsleep example](https://gist.github.com/dedalodaelus/159b73a036c250267d0dbbf496eb96f6)

```yaml
---
substitutions:
  devicename: esp32-arduino
  friendly_name: Dev Full Board
  sleep_time: 20s
  wifi_signal_update: 20s
  #sensors_update: 1s
  uptime_update: 2s
  
esphome:
  name: $devicename
  platform: ESP32
  board: esp32doit-devkit-v1
  on_boot:
    then:
      - deep_sleep.prevent: deep_sleep_control
  on_shutdown:
    then:
      - lambda: |-
         id(ts_saved) = id(homeassistant_time).now().timestamp;

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_psk
  
  fast_connect: true
  reboot_timeout: 3min
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "hotspot"
    password: "1234567890"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

globals:
  - id: first_time_boot
    type: bool
    initial_value: "true"
    restore_value: yes
  - id: uptime_saved
    type: time_t
    initial_value: "0"
    restore_value: yes
  - id: ts_saved
    type: time_t
    initial_value: "0"
    restore_value: yes
  - id: wifi_signal_sent
    type: bool
    initial_value: "false"
    restore_value: no
  - id: uptime_sent
    type: bool
    initial_value: "false"
    restore_value: no
  - id: max_disconnect_time
    type: time_t
    initial_value: "30"
    restore_value: no
  - id: tsync_read
    type: bool
    initial_value: "false"

time:
  - platform: homeassistant
    id: homeassistant_time
    on_time_sync:
      then:
        - script.execute: calculate_uptime

deep_sleep:
  id: deep_sleep_control
  sleep_duration: $sleep_time

binary_sensor:
  - platform: homeassistant
    id: prevent_deep_sleep
    name: $friendly_name Prevent Deep Sleep
    entity_id: input_boolean.prevent_deep_sleep
  - platform: status
    id: status_sensor
    name:  $friendly_name Status

sensor:
  - platform: wifi_signal
    name: $friendly_name WiFi Signal
    on_value:
        then:
          - lambda: |-
              id(wifi_signal_sent) = true;
          - script.execute: consider_deep_sleep
    update_interval: $wifi_signal_update
    filters:
      - heartbeat: 3s
  - platform: uptime
    id: uptime_sensor
    name: $friendly_name Uptime
    update_interval: $uptime_update
    filters:
      - lambda: |-
          if (
            !id(homeassistant_time).now().is_valid() 
            &&
            !id(tsync_read)
          ) {
            ESP_LOGD("uptime", "time-not-valid");
            return id(uptime_sensor).raw_state;
          }
          time_t time_now = id(homeassistant_time).now().timestamp;
          // ESP_LOGD("uptime", "prev ts %d", id(ts_saved));
          time_t uptime_delta = time_now-id(ts_saved);
          ESP_LOGD("uptime", "uptime delta: %d", uptime_delta);
          if (uptime_delta > id(max_disconnect_time)) {
            ESP_LOGD("uptime", "Reseting uptime");
            id(uptime_saved)=0;
          }
          else {
            id(uptime_saved)+=uptime_delta;  
          }
          if (id(uptime_saved)<0)
          {
            ESP_LOGD("uptime", "Something went wrong negative uptime");
            id(uptime_saved)=0;
          }
          id(ts_saved) = time_now;  
          // ESP_LOGD("uptime", "time now %d", time_now);
          id(uptime_sent) = true;
          return id(uptime_saved);

light:
  - platform: binary
    name: $friendly_name Internal led
    output: internal_led_output

output:
  - id: internal_led_output
    platform: gpio
    pin: GPIO2

script:
  - id: calculate_uptime
    mode: queued
    then:
      - if:
          condition:
            time.has_time:
          then:
            - logger.log: "Saving timestamp"
            - lambda: |-
                time_t time_now = id(homeassistant_time).now().timestamp;
                // ESP_LOGD("uptime", "prev ts %d", id(ts_saved));
                time_t uptime_delta = time_now-id(ts_saved);
                ESP_LOGD("uptime", "uptime delta: %d", uptime_delta);
                if (uptime_delta > id(max_disconnect_time)) {
                  ESP_LOGD("uptime", "Reseting uptime");
                  id(uptime_saved)=0;
                }
                id(ts_saved) = id(homeassistant_time).now().timestamp;
                id(tsync_read) = true;

  - id: consider_deep_sleep
    mode: queued
    then:
      - logger.log: "Considering Deep Sleep"
      - delay: 5s
      - if:
          condition:
            and:
              - wifi.connected
              - api.connected
              - binary_sensor.is_off: prevent_deep_sleep
              - lambda: |-
                  if (
                    id(uptime_sent) &&
                    id(wifi_signal_sent)
                  ) {
                    ESP_LOGD("sensors", "All sensors data sent");
                    return true;
                  }
                  return false;
          then:
            - logger.log: "Entering in Deep Sleep"
            - deep_sleep.enter: deep_sleep_control
          else:
            - logger.log: "Skipping Deep Sleep"
            - delay: 1s
```