esphome/feature-requests #1082: SIM800L: can not read RSSI on HASSIO with ESPHome

fahhome123, 2021-01-11:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 with SIM800L

**ESPHome version (latest production, beta, dev branch)**

ESPHome 1.15.3

**Affected component:**
https://esphome.io/components/sim800l.html

**Description of problem:**
ESP32 with SIM800L can send and receive message. This works. 
I use it on HomeAssistant for security things.
I want to trigger the RSSI from SIM800L, but it not works.

RSSI is not readable.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

fahhome123, 2022-02-16:
Hello, any solution or date when ist comes?

glmnet, 2022-04-14:
Closing this as the rssi was implemented about two releases ago

closed 2022-04-14 by glmnet