esphome/feature-requests #1083: M5Paper support

megawubs, 2021-01-12:
**Describe the problem you have/What new integration you would like**

I've got my hands on the M5Paper from M5Stack and would love to use the e-ink touch display to render switches and buttons that communicate with HomeAssistant through ESPHome. I have no idea if this is the correct place to put this request, as I'm fairly new to programming micro controllers

**Please describe your use case for this integration and alternatives you've tried:**

I've searched the documentation of ESPHome to find a way to interacht with the touch screen and e-ink display of the M5Paper, but I could not find something useful or something that worked. It might be my own shortcoming, as I do not have that much experience with ESPHome or ESP32 in general.

**Additional context**

It might be noteworthy that M5 already has their own library [M5EDP](https://github.com/m5stack/M5EPD) to do basic stuff with the M5Paper. This might be a starting point? The link to the Github M5EPD project also contains a bit more information about the device and the hardware configuration.

megawubs, 2021-01-13:
I would love to help out in any way, as I'm a programmer myself. But the problem is that I don't have experience with C/C++. I've tried some things, but a lot is unclear to me of how it works. I think that with a little help form someone more experienced I would be able to provide a PR to add support for this device.

tallnhairydave, 2021-01-14:
Gets a big thumbs up from me. Mine just arrived and would love to be able to tie it in with ESPHOME and HA. It's a nice bit of. kit :) 

Here's the M5 docs on the APIs and relevant functions for the device if they help at all? 

- display - https://docs.m5stack.com/#/en/api/m5paper/epd_canvas
- touch - https://docs.m5stack.com/#/en/api/m5paper/touch
- system button, RTC and onboard SHT30 sensor - https://docs.m5stack.com/#/en/api/m5paper/system_api

The touch screen is an i2c device, using the Goodie GT911 controller https://www.distec.de/fileadmin/pdf/produkte/Touchcontroller/DDGroup/GT911_Datasheet.pdf

Sadly, like megawubs, I've not got much C/C++ experience

megawubs, 2021-01-17:
Ive been trying to get a sample ESPHome project to compline with the M5EPD project as a custom library, but this gives me too much errors that I cannot resolve. It's complaining about a lot of unknown references. Especially the SD reference. I've forked the M5EPD library and tried to remove the SD stuff in there but that's a too deep rabbit hole for me atm. 

I think a better approach for me right now is to just work with the M5EPD library in combination with the M5Paper UI library and publish/subscribe to mqtt stuff. It seems it's too difficult to just "merge" the two projects together.

cyrilpawelko, 2021-01-21:
I'm working myself on a m5paper panel, for openhab and without esphome.
I already have a working proof-on-concept on https://github.com/cyrilpawelko/m5panel
It's not complete (no touch support for example), but can be used as a starting point.

fonix232, 2021-01-22:
@megawubs it will give you errors since the M5EPD project relies on the Arduino libraries. Take a look at the Waveshare Epaper component for examples on what to replace with what. E.g. the M5EPD library uses this include for SPI support:
```
#include <SPI.h>
```

However on ESPHome, you need to use this:

```
#include "esphome/components/spi/spi.h"
```

I'd also not hold my breath for SD card support.

Also, there's a lot of work to be done on the separate components of the M5Paper to make it completely usable. Battery voltage reporting would be nice to have (might be tough due to the different ADC setup in M5EPD), touchscreen is probably doable, but you won't be able to do much with it... You'll need to supply your own STH30 module as well.

@cyrilpawelko I'm not sure that your PoC will be of any use, since it relies on the Arduino framework fully. ESPHome provides its own framework, with different includes, different available packages, and so on... Porting everything wouldn't be an easy task, and I'm fairly certain the developers wouldn't want to pull in so many extra resources. The point of ESPHome is to create components that integrate with HomeAssistant, and most of the stuff is not meant to run on its own. The M5Paper's full support requires a level of independent functionality that would be way out of scope for this project.

megawubs, 2021-01-24:
@fonix232 thanks for your reply. This is what I was afraid of. I think I have to wait for M5Stack to fully support this device with UIFlow and micropython. 

That said, do you know if it's possible to get the home assistant protocol esphome uses to work outside of esphome? If I can include that within the M5EPD project it would be a great step forward. I've been trying with an home assistant rest client, but that's a big hassle to get working correctly when you use https.

jhoult, 2021-01-28:
@megawubs according to m5stack, you need to use the "beta" selection when you start uiflow.  I just downloaded it (just found my m5paper in the mailbox half an hour ago) and confirmed it's in there.  havent touched micropython yet.

megawubs, 2021-01-28:
@jhoult I'm not seeing it when I open UIFlow 1.7.2 beta :( 

[image]

sj3fk3, 2021-02-06:
Would also love to see the M5Paper supported by esphome.

eecue, 2021-04-07:
Following as I'd love some support here.

JasonSwindle, 2021-04-13:
+1 to this, so I can use home assistant with this.

richardklingler, 2021-05-17:
Have a few m5papers here, but haven't started yet as it is not on my top priority list....

https://github.com/richardklingler/esphome

HElkhoury, 2021-08-01:
I'd add my support for this too

jravetch, 2021-09-09:
Same here.  M5Paper seems like a perfect fit for esphome.

ngelotte, 2021-11-23:
This would be a great addition to home assistant.

Lexorius, 2022-01-20:
m5paper push ... ;) would like to use the m5paper as a remote would be perfekt for this

teun95, 2022-03-18:
Such a neat looking device. Support would be great!

Lexorius, 2022-04-17:
is there something new ? (push - again, because the device is realy nice)

superinj, 2022-05-09:
+1, would be great!

sebirdman, 2022-08-11:
I've started bringing up the individual peripherals. I think the display will be the most challenging. The rest are somewhat simple i2c devices. 

I'll post a git repo with some external components as soon as it's in a better state.

Yoneo, 2022-08-13:
Does anyone know if it is possible to do refreshes when the M5Paper is not plugged in? 

I want to use it as weather station. But it seems that I have to press the on button every time or let it permanently connected (see also https://github.com/Bastelschlumpf/M5PaperWeather/issues/1). Thanks in advance for the answers!

Yoneo, 2022-09-03:
> Does anyone know if it is possible to do refreshes when the M5Paper is not plugged in?
> 
> I want to use it as weather station. But it seems that I have to press the on button every time or let it permanently connected (see also [Bastelschlumpf/M5PaperWeather#1](https://github.com/Bastelschlumpf/M5PaperWeather/issues/1)). Thanks in advance for the answers!

Maybe this is an answer: https://github.com/Bastelschlumpf/M5PaperWeather/issues/1#issuecomment-1220284778 (not tested yet)

mmccool, 2022-09-09:
So I am very interested in this to, to use as a control panel stuck to my fridge.  In general it would be nice to see more support for eInk touch displays to make control panels with.   BTW I tried to use a similar kit for this purpose, the Adafruit Magtag, but that uses an ESP-S2... which is apparently not supported by ESPHome, another annoyance.  I got an inkplate (which I have not tried setting up yet) but it's pretty basic and the unit I got had no GPIO (parts shortage...).

My short term plan is to set it up as an MQTT device (using either direct arduino programming or uiflow) but... ESPHome would be nicer, I have a bunch of other ESP devices set up that way already.

sebirdman, 2022-09-10:
https://github.com/sebirdman/m5paper_esphome

Things are generally functional. There's some interesting changes that will have to happen to make the esphome spi component work with the display, so there's a forked version in this repo. 

At some point, will work to get it upstreamed.

k-w-1, 2023-03-30:
I've also been keeping a close eye and unfortunately changes in all 10 forks don't seem particularly significant from sebridman's.

That said, I've been able to get it to compile & work, however I had to hard code a bunch of stuff in IT8951ESensor::get_device_info. It seems like my device (nb: I have the "Community" edition with the semi-translucent case) returns garbage values for the screen H & W; the subsequent malloc calls end up crashing things as it tries to allocate way too much RAM. Has this happened to anyone else? It ends up causing a watchdog reset (until it falls back to ESPHome safe mode, anyway).

error33-priv, 2023-07-26:
@k-w-1 I am having the same issue with the screen H & W values

Dominaezzz, 2023-08-15:
When reading the device info from the M5Paper you have to do it slowly, so during initialisation the spi frequency needs to be lower, about 1MHz has been safe for me. After getting the device info you can ramp up to 10MHz (or even 20MHz iirc).

AndBu, 2023-10-29:
Shamelessly bumping this thread. Native M5Paper EspHome integration would be incredible!

sepinho, 2023-11-19:
Anything I can do other than coding to speed things up? I'd be willing to donate some (double-digit) funds to get M5paper into ESPHome. I really, really want to turn it from a paperweight to a universal remote, taking some inspiration from Sony's Huis.

hgn32, 2023-11-28:
> When reading the device info from the M5Paper you have to do it slowly, so during initialisation the spi frequency needs to be lower, about 1MHz has been safe for me. After getting the device info you can ramp up to 10MHz (or even 20MHz iirc).

How should I code to lower the spi frequency to 1MHz?
to m5paper-esphome/it8951e/it8951e.cpp

Dominaezzz, 2023-11-28:
> How should I code to lower the spi frequency to 1MHz?

Unfortunately I have never used esphome so I have no idea.

Dominaezzz, 2023-11-28:
I would change the frequency.

hgn32, 2023-11-29:
> Then what would you change in https://github.com/idefxH/m5paper_esphome/blob/9afd44eab19d714bfaca22da513936c200f0b678/components/it8951e/it8951e.cpp to do it?

I changed this, but it didn't improve. P.I.
https://github.com/idefxH/m5paper_esphome/blob/9afd44eab19d714bfaca22da513936c200f0b678/components/it8951e/it8951e.h#L13C56-L13C56

Dinth, 2023-12-21:
Bumping this up as gt911 support was added in the latest release of ESPHome.
Keeping fingers crossed that soon we will see M5Paper fully supported

sepinho, 2023-12-21:
Yeah, that caught my eye as well. Here's to hoping!

fonix232, 2023-12-29:
I'm actually working on rewriting the IT8951 driver from the repos listed above - so far I've managed to work around many of the hacky bits, and am able to clear the screen. It is a bit finicky because all the drivers above are... Well, not well implemented, at least not according to [the spec](https://www.waveshare.net/w/upload/1/18/IT8951_D_V0.2.4.3_20170728.pdf).

I'm somewhat busy at the moment (holidays + jobhunt), but I'll be pushing the custom component in a while.

Passific, 2024-01-11:
Hello everyone, I also spend some time on the topic, and I think I now have something stable.
Some part could be better, some less M5EPD specific but most importantly it works with latest ESPHome 2023.12.5!

Some driver (gt911) are now part of the official repo, specific SPI was not needed neither so I remove them both.

Enjoy your M5Paper: https://github.com/Passific/m5paper_esphome

Stoppi7, 2024-04-12:
Du musst zu Beginn die Taste auf der Rückseite drücken, damit das Display nicht mehr eingefroren ist. Bei erneutem Drücken wird das Display wieder eingefroren.

EpicLPer, 2024-05-28:
> Hello everyone, I also spend some time on the topic, and I think I now have something stable. Some part could be better, some less M5EPD specific but most importantly it works with latest ESPHome [...]

Planning on buying an M5Paper and doing some research beforehand now, does this also include battery saving functionalities? I'm not "too" familiar with ESPHome, so I'm not entirely sure how I'd go about this :) But my current understanding is I'd have to implement my own saving function somehow?

ZeFish, 2024-09-04:
It still have issue with the busy pin that trigger reboot. There is a fork tho that resolve some busy pin issue but it’s still present
https://github.com/timit3-github/m5paper_esphome