esphome/feature-requests #1084: Add support for TFT display ST7796S

morganrallen, 2021-01-12:
**Describe the problem you have/What new integration you would like**

Add support for the TFT display ST7796S
**Please describe your use case for this integration and alternatives you've tried:**

First attempts at using the ST7789V display driver didn't work and two things of note came up.
* This (all?) display drivers require a reset pin, but in some cases (TTGO-T-WATCH) this isn't wired up, maybe as is usually the case in the ESP-IDF, accept `-1` to mean not in use
* Display drivers seem to hard code dimensions, making this more flexible may help with some of these very similar but not quite displays.

**Additional context**

JingleheimerSE, 2021-03-18:
I would also like to see a ST7789 driver added. This would intergrate well with the Seeed WT32-SC01 board that integrates a 3.5" ST7789S display with an ESP32 for only 20 USD!

[https://www.seeedstudio.com/ESP32-Development-board-WT32-SC01-p-4735.html](https://www.seeedstudio.com/ESP32-Development-board-WT32-SC01-p-4735.html)

richardklingler, 2021-05-17:
The WT32-SC01 uses the ST7796S and not ST7789...

[image]

morganrallen, 2021-06-14:
That's great news @richardklingler but unless I'm missing something, I don't see any display drivers in that repo.

jamser, 2021-06-24:
That's great @richardklingler but i cloud not open the link you share,cloud you send it to me(esphome.zip), my mail is szzs80@163.com,thank you very much!~

cpyarger, 2021-06-25:
And now that repo is gone :/

kd7gab, 2021-11-10:
I would really like to be able to utilize the WT32-SC01 units with ESPHome. Has this request simply been abandoned?

cpyarger, 2022-01-22:
It seems that @richardklingler has reposted the github repository but is putting the devices he got working behind a sponsor/pay wall.

cpyarger, 2022-07-01:
Please post the relevant information prior to closing. What external component, where is the link to the code,  documentation, examples, etc.

nagyrobi, 2022-07-01:
reopening as requested on Discord