esphome/feature-requests #1085: Allow buad rate to be change on run command

morganrallen, 2021-01-12:
**Describe the problem you have/What new integration you would like**

There is no option to change the default `115200` baud rate for `esptool` even though many devices support much faster rates. Hopefully this is a simple case of adding a switch and passing that to `esptool`

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

morganrallen, 2021-01-12:
I found a work around to this. Setting `esphome.platformio.upload_speed`.

But not only did it take me digging through the code to find this, as this is an `esptool` option, using the `platformio` key is confusing.

glmnet, 2021-01-13:
it's suggested here to do just that https://deploy-preview-924--esphome.netlify.app/components/esphome.html#esphome-platformio-options

We all do it that way, what do you suggest as an improvement?

morganrallen, 2021-01-13:
ah, as I'm not using platformio I had missed that part of the documentation.

my suggestion would be two fold:
* add `esptool_options` to the `esphome` key, expose all the options(?)
* add `--upload-baud` switch to the `esphome` command to compliment the `--upload-speed` switch.

I'd probably prioritize that way too... CLI overrides config, which overrides default.