esphome/feature-requests #1087: ATM90e32 Platform - Sensors on Separate Chips not Updating/Publishing at the Same Time. So Calculated Sensor Values are Incorrect.

Teleportist, 2021-01-14:
_**Describe the problem you have/What new integration you would like**_

ATM90e32 ESPHome platform.

I am using the ATM90E32AS based board from CircuitSetup to monitor power on 6 circuits.
[https://circuitsetup.us/index.php/product/expandable-6-channel-esp32-energy-meter/](url)

The 2 Microchip ATM90E32AS are measuring input current from 3 separate channels each.
Unfortunately it seems the data from each chip is published at different times.~ 20 seconds or so apart.
This means that if I need to include a virtual/calculated sensor, based on measurements from channels on separate chips
there is a chance the readings will no be aligned, so producing an incorrect result.

REQUEST IS: An option to synchronise the output from form both chips, to force them to publish at the same time.

_**Please describe your use case for this integration and alternatives you've tried:**_

EXAMPLE:
Hopefully the below screen grab from Grafana will clearly show the issue.
https://user-images.githubusercontent.com/61894621/103577465-caf0b680-4ecc-11eb-8027-a8f88060f09d.png

'Total' (CT1) = Incoming cable from the electricity grid
'Ground & 1st Floors' (CT6) = Cable into primary fuse board
'2nd Floor' (calculated) = Total(CT1) - Ground & 1st Floors (CT6)*

*Should be what my secondary fuse board, for the 2nd floor, is consuming.

As you can see, the calculated '2nd Floor' sensor (purple) is reporting negative values, because the 'Total' sensor (green) is lagging behind the 'Ground & 1st Floors' sensor. For the calculation to be accurate the readings should be from the same point in time.

So if CT1 is 300W when it updates, but CT6 hasn't updated from it's last value, of say 500W, my calculated sensor will be -200W (300W-500W). Then after CT6 updates just after with the new value of 200W, so the calculated sensor value should have been 100W (300W-200W). But now it's too late.

I don't have a spare channel to actually clamp on to the cable into the secondary fuse board, which services the second floor.

I have since moved both CT clamps on to the same chip, which has fixed the issue, but I'm planning to buy further boards, to add more channels, where again the 2 new chips would publish out of step with those on the other board, as well as each other.

**Additional context**

Here is my YAML.
`tutions:
  disp_name: PowMon6C
  update_time: 60s
# Current Transformers:
#  20A/25mA SCT-006: 11143
#  30A/1V SCT-013-030: 8650
#  50A/1V SCT-013-050: 15420
#  80A/26.6mA SCT-010: 41660
#  100A/50ma SCT-013-000: 27518
#  120A/40mA: SCT-016: 41787
#  200A/100mA SCT-024: 27518
  #current_cal: '27518'
  current_cal: '28304'
# Jameco 9VAC Transformer: 
#  For meter versions: 
#  >= v1.3: 7305
  #voltage_cal: '7305'
  voltage_cal: '13611'

esphome:
  name: power_mon_6c
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

# Enable logging
logger:

# Enable Home Assistant API
#api:
mqtt:
   broker: !secret mqtt_broker
   username: !secret mqtt_user
   password: !secret mqtt_password

ota:
  password: !secret ota_password

web_server:
  port: 80

spi:
  clk_pin: 18
  miso_pin: 19
  mosi_pin: 23

sensor:
  - platform: wifi_signal
    name: ${disp_name} WiFi
    update_interval: 60s
#IC1
  - platform: atm90e32
    cs_pin: 5
#Incomming
    phase_a:
      voltage:
        name: ${disp_name} Volts A
        id: ic1Volts
        accuracy_decimals: 1
      current:
        name: ${disp_name} CT1 Amps
        id: ct1Amps
# The max value for current that the meter can output is 65.535. If you expect to measure current over 65A, 
# divide the gain_ct by 2 (120A CT) or 4 (200A CT) and multiply the current and power values by 2 or 4 by uncommenting the filter below
#        filters:
#          - multiply: 2
      power:
        name: ${disp_name} CT1 Watts
        id: ct1Watts
#        filters:
#          - multiply: 2
        filters:
          - multiply: -1
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
#Shed
    phase_b:
      current:
        name: ${disp_name} CT2 Amps
        id: ct2Amps
      power:
        name: ${disp_name} CT2 Watts
        id: ct2Watts
        filters:
          - multiply: -1
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
#GF & FF
    phase_c:
      current:
        name: ${disp_name} CT3 Amps
        id: ct3Amps
      power:
        name: ${disp_name} CT3 Watts
        id: ct3Watts
        filters:
          - multiply: -1
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
    frequency:
      name: ${disp_name} Freq A
    line_frequency: 50Hz
    gain_pga: 1X
    update_interval: ${update_time}
#IC2
  - platform: atm90e32
    cs_pin: 4
    phase_a:
#this voltage is only needed if monitoring 2 voltages
#      voltage:
#        name: ${disp_name} Volts B
#        id: ic2Volts
#        accuracy_decimals: 1
#GF Lighting
      current:
        name: ${disp_name} CT4 Amps
        id: ct4Amps
      power:
        name: ${disp_name} CT4 Watts
        id: ct4Watts
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
#GF Sockets
    phase_b:
      current:
        name: ${disp_name} CT5 Amps
        id: ct5Amps
      power:
        name: ${disp_name} CT5 Watts
        id: ct5Watts
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
#Kitchen
    phase_c:
      current:
        name: ${disp_name} CT6 Amps
        id: ct6Amps
      power:
        name: ${disp_name} CT6 Watts
        id: ct6Watts
        filters:
          - multiply: -1
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
    frequency:
      name: ${disp_name} Freq B
    line_frequency: 50Hz
    gain_pga: 1X
    update_interval: ${update_time}
        
#Total Amps   
#  - platform: template
#    name: ${disp_name} Total Amps
#    id: totalAmps
#    lambda: return id(ct1Amps).state + id(ct2Amps).state + id(ct3Amps).state + id(ct4Amps).state + id(ct5Amps).state + id(ct6Amps).state ;
#    accuracy_decimals: 2
#    unit_of_measurement: A
#   icon: "mdi:flash"
#    update_interval: ${update_time}         
#Total Watts
#  - platform: template
#    name: ${disp_name} Total Watts
#    id: totalWatts
#    lambda: return id(ct1Watts).state + id(ct2Watts).state + id(ct3Watts).state + id(ct4Watts).state + id(ct5Watts).state + id(ct6Watts).state ;
#    accuracy_decimals: 1
#    unit_of_measurement: W
#    icon: "mdi:flash-circle"
#    update_interval: ${update_time}
#kWh
  - platform: total_daily_energy
    name: ${disp_name} Total kWh
    power_id: ct1Watts
    filters:
      #- multiply: -1
      - multiply: 0.001
    unit_of_measurement: kWh
#Top Floor Amps   
  - platform: template
    name: ${disp_name} Loft Amps
    id: loftAmps
    lambda: return id(ct1Amps).state - id(ct3Amps).state ;
    accuracy_decimals: 2
    unit_of_measurement: A
    icon: "mdi:flash"
    update_interval: ${update_time}
#Top Floor Watts
  - platform: template
    name: ${disp_name} Loft Watts
    id: loftWatts
    lambda: return id(ct1Watts).state - id(ct3Watts).state ;
    accuracy_decimals: 1
    unit_of_measurement: W
    icon: "mdi:flash-circle"
    update_interval: ${update_time}

switch:
  - platform: restart
    name: ${disp_name} Restart  
time:
  - platform: sntp
    id: sntp_time`

Teleportist, 2022-05-19:
Just checking in, after a year, to bump this in the hope it gets assigned. Thanks

CircuitSetup, 2022-09-30:
You may want to try changing the update_interval for the meter to less than the template calculation. This way, the update is staggered a bit. 

You may want to look at why CT6 is reading negative though. I see you put a multiply -1 in there - was it reading negative most of the time, previously? This looks like the CT is backwards on the wire, which would be giving you an incorrect power factor reading, calculating an incorrect power (watts) in the meter.

Cougar, 2023-01-04:
I'm thinking if it is possible to disable normal updates (with `update_interval: 0` ?) and then run script with timer using lambdas to call `ATM90E32Component::update()` for both ICs sequentially? This could be quite easy to implement I guess.

RamyaDidde, 2024-02-13:
hi,

i am using ATM90E32 IC, i am facing issue while reading Kwh from APenergyT register. Can anyone tell how to read Kwh and any calibration register to improve Kwh.
Thanks in advance.

CircuitSetup, 2024-02-13:
> hi,
> 
> i am using ATM90E32 IC, i am facing issue while reading Kwh from APenergyT register. Can anyone tell how to read Kwh and any calibration register to improve Kwh. Thanks in advance.

I assume you're using the instructions for the config file here: https://esphome.io/components/sensor/atm90e32#active-energy ?

Each phase must be read, as APenergyT (the total register) is not used in ESPHome.

Also, the output is in watt hours. So when these registers are read, and the data is stored, the time between reads must be taken into account since they are accumulation registers

CircuitSetup, 2024-02-13:
> I'm thinking if it is possible to disable normal updates (with `update_interval: 0` ?) and then run script with timer using lambdas to call `ATM90E32Component::update()` for both ICs sequentially? This could be quite easy to implement I guess.

There's an update coming that mitigates time between reading the registers and publishing. https://github.com/esphome/esphome/pull/5670