esphome/feature-requests #1091: Flash alternative firmware again (e.g. tasmota) using esphome's OTA mechanism

pfichtner, 2021-01-17:
**Describe the problem you have/What new integration you would like**

The docs states how to migrate from [tasmota to esphome](https://esphome.io/guides/migrate_sonoff_tasmota.html). Once esphome is working is there a way to upload another firmware than esphome itself (e.g. tasmota again) using esphome's ota mechanism? 

**Please describe your use case for this integration and alternatives you've tried:**

Searched the web ui and [command line args](https://esphome.io/guides/cli.html) but could not find anything describing this scenario. 

**Additional context**

It's not clear if esphome supports this feature already and it's just missing in the docs or if it's not implemented (yet). Would be nice if this is possible because I once did flash tasmota to my sonoff relay then migrated to esphome (via tasmota) and if there is no way back to another firmware I would have to open the case again for soldering pins for flashing using usb.

marcinkowalczyk, 2021-01-17:
https://esphome.io/components/ota.html ?

pfichtner, 2021-01-17:
Yes, forgot to mention: https://esphome.io/components/ota.html also does not provide information how to achieve it.

marcinkowalczyk, 2021-01-17:
Just mix it together with https://esphome.io/components/captive_portal.html :)

pfichtner, 2021-01-17:
That's it! Thanks! Using the captive portal's fallback mechanism there's a way to specify an alternative firmware file. Perhaps this could be stated more clearly in the docs? 
So if agree that this could be documented a little bit better leave the issue open otherwise it could be closed.

pfichtner, 2021-01-17:
It would be nice if there's a parameter for the CLI where an alternative firmware could be passed. The only working process at the moment is: 
- Add captive portal support to the firmware and reflash the device (including wrong wifi credentials so that esp can't connect to the real wifi and captive portal switches to the fallback)
- wait for the SSID (fallback AP) to appear and connect to it
- upload the alternative firmware using a browser

Not only this is more complicated that it could be there a devices where the flash is not big enough to include the captive portal. So yes, there's a walk around for my problem but no real feature.

closed 2022-07-01 by nagyrobi