esphome/feature-requests #1092: Support for CM1106, CP1107

SangzinAhn, 2021-01-20:
**Describe the problem you have/What new integration you would like**

I would like ESPHome to support the NDIR CO2 sensors CM1106 and CM1107 from Cubic.
https://en.gassensor.com.cn/CO2Sensor/list.html

**Please describe your use case for this integration and alternatives you've tried:**

I am still learning Arduino language, so I was not able to try anything yet.

**Additional context**

Thank you for the amazing ESPHome add-on. This makes my Smart home project very flexible.
I would like to use the NDIR CO2 sensor CM1107 in order to monitor indoor air quality and automate ventilation accordingly.
CO2 sensors tend to drift over time, and usually needs to be calibrated time to time.
However, it states that CM1107 uses the dual beam method, hence the drift is negligible.
I think it will be a good addition the list of ESPHome supported CO2 sensor list.

I have attached the datasheet. It says it supports UART, I2C, PWM as signal output.
[Single Beam NDIR CO2 Sensor Module CM1106 Specification.pdf](https://github.com/esphome/feature-requests/files/5839487/Single.Beam.NDIR.CO2.Sensor.Module.CM1106.Specification.pdf)
[Dual Beam NDIR CO2 Sensor Module CM1107 Specification.pdf](https://github.com/esphome/feature-requests/files/5839488/Dual.Beam.NDIR.CO2.Sensor.Module.CM1107.Specification.pdf)

LeoDJ, 2021-02-02:
I wrote a custom sensor component for the CM1106 sensor using the UART protocol.

You can find the code here: https://github.com/LeoDJ/ESPHome_Nodes/tree/main/CM1106_CO2_Sensor

I hope this helps ;)

SangzinAhn, 2021-02-02:
Oh my! Thank you for your great work. I will dive in!

LeoDJ, 2021-02-02:
No problem, but I think you can keep this issue open. Native support for the sensor would be great.

I just don't think my code would be robust enough to contribute to ESPHome directly. Otherwise I'd make a pull request.

SangzinAhn, 2021-02-03:
Thank you for your advice. I will reopen this issue. Sorry if this is some sort of spamming.

SangzinAhn, 2021-02-06:
Quick feedback to @LeoDJ.  I applied your code to my CM1107 and it works just fine. Thanks once again!

nagyrobi, 2022-07-01:
Feel free to submit your component to ESPHome as a PR.

closed 2022-07-01 by nagyrobi