esphome/feature-requests #1096: GOSUND LB3

pepsonEL, 2021-01-22:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

jkolczasty, 2021-12-29:
Hi
I just received this device today and did some tests and I have good news for you (and probably others).

Status so far:
1. It's ESP8266EX
2. It's not convertible by tuya-convert
3. PCB has serial pinouts on hidden side (need to remove pcb from device for first flashing)
4. Flashed with esphome using serial sucessfully (sample config below)
5. GPIO:
       GPIO13 = WLED PWM
       GPIO12 = G_LED PWM
       GPIO5 = B_LED PWM
       GPIO15 = R_LED PWM

TODO:
There is i2c touch sensor for side touch "buttons", it's On-Chip IC. Haven't found any info about it yet.

```yaml
substitutions:
  devname: mynitebird

output:
  - id: "${devname}_output_wled"
    platform: esp8266_pwm
    pin: GPIO13
    frequency: 1000 Hz

  - id: "${devname}_output_gled"
    platform: esp8266_pwm
    pin: GPIO12
    frequency: 1000 Hz

  - id: "${devname}_output_bled"
    platform: esp8266_pwm
    pin: GPIO5
    frequency: 1000 Hz

  - id: "${devname}_output_rled"
    platform: esp8266_pwm
    pin: GPIO15
    frequency: 1000 Hz

light:
  - platform: monochromatic
    name: "${devname}_wled"
    id: "${devname}_wled"
    output: "${devname}_output_wled"

  - platform: monochromatic
    name: "${devname}_gled"
    id: "${devname}_gled"
    output: "${devname}_output_gled"

  - platform: monochromatic
    name: "${devname}_bled"
    id: "${devname}_bled"
    output: "${devname}_output_bled"

  - platform: monochromatic
    name: "${devname}_rled"
    id: "${devname}_rled"
    output: "${devname}_output_rled"
```

ams2121, 2022-01-10:
@jkolczasty

Thanks for what you posted above, I appreciate your work.  I wanted to share back what added progress I've been able to make (though I'm a Tasmota user).  The I2C bus is on the following GPIOs:

GPIO2 - I2C SDA
GPIO14 - I2C SCL

I was able to query what is (I think) both bus 1 and bus 2 and found devices at address 0x39 and 0x45.  This does not explicitly apply to a specific driver that I've found yet, I guess I need to pop the bottom off the light again and try to read the super tiny chip again (arg!).

If I figure out the chip or manage to talk to it, I'll post some more.  

If it is of use to you, in trying to tune the color gamma, I found it was very weak on the red, a little weak on green, and that the white LED was very bright (when transitioning between color and white light.  I can share that if it helps you, though not sure if how to translate between Tasmota and EspHome.

All the best.

jkolczasty, 2022-01-10:
It's KF8TS2716 chip. It's very same what is used in xiaomi lamps and someone already did some research on that:

https://github.com/mmakaay/esphome-xiaomi_bslamp2/blob/d97c3361b09dacbab0dc9452bdd3c3e1e1235967/doc/technical_details.md

Currently have no time to write some test code for esphome, but it's still on top of my to-do.txt file ;-)

ams2121, 2022-01-10:
I just found that same/slightly link!  (https://github.com/mmakaay/esphome-xiaomi_bslamp2/blob/dev/doc/technical_details.md) You are totally on top of this.   I've never written an i2c driver, though thinking of writing one also.  I'll put a link to code here as I play with it (hopefully it's only somewhat awful!)

flixlix, 2022-03-18:
Are there any news on this? Has anyone gotten the touch function to work?

jkolczasty, 2022-03-21:
Had not time to look at it so far. Maybe next week will find some time ;-)

bfranske, 2022-09-09:
It's unclear what the alternative offered is here so I'm not sure why this was closed.

ams2121, 2022-09-10:
I'm using mine, remote only via MQTT (with Tasmota).  Getting the touch to work would be nice, however its definitely beyond my skill level.  @jkolczasty is the real pro!  anyhow best to all.

pepsonEL, 2022-11-08:
Aby new about touch ?

Ecco, 2024-02-08:
So, I did reverse engineer those lamps, and got both the touch and the tap-to-turn-on features to work! The "tap-to-turn-on" is handled by a cheap MEMS accelerometer, whose support I submitted in #6187. The great news is that the behavior can be heavily customized (single tap, double tap, length, intensity), *and* you get your lamp's acceleration as a bonus (extremely useful ofc 🤣).

The touch part (at least on my model), is (poorly) handled by a `cbm7320` IC chip that talks over I2C. This chip is not very well put together : if the lamp is above 60% brightness the chip's touch sensing mechanism gets confused and the chip start spitting out nonsense on I2C… Also that chip is in fact a general purpose MCU with a specific firmware made to transmit touch sensing events over I2C, so I don't know if a driver makes sense for esphome.

Anyway, here's my custom config (just ignore the cbm7320 part for now). But with #6187 the da270 part works well 😄 

```yaml
esphome:
  friendly_name: "Bedside Lamp"

esp8266:
  board: esp01_1m

output:
  - platform: esp8266_pwm
    id: blue_led
    pin: GPIO5
  - platform: esp8266_pwm
    id: green_led
    pin: GPIO12
  - platform: esp8266_pwm
    id: warm_white_led
    pin: GPIO13
  - platform: esp8266_pwm
    id: red_led
    pin: GPIO15

light:
  - platform: rgbw
    id: main_light
    name: "Light"
    red: red_led
    green: green_led
    blue: blue_led
    white: warm_white_led

i2c:
  sda: GPIO2
  scl: GPIO14
  scan: false
  frequency: 32 kHz # Matches the default firmware

external_components:
  - source: components

cbm7320:

binary_sensor:
  - platform: gpio
    icon: 'mdi:gesture-tap-box'
    pin:
      number: GPIO4
      mode:
        input: true
        pullup: false
    name: "Tap Detector"
    on_press:
      - light.toggle: main_light
  - platform: cbm7320
    name: Power Button
    channel: 14
    on_press:
      - light.toggle: main_light
  - platform: cbm7320
    name: Left Slider A
    channel: 7
  - platform: cbm7320
    name: Left Slider B
    channel: 8
  - platform: cbm7320
    name: Left Slider C
    channel: 9
  - platform: cbm7320
    name: Left Slider D
    channel: 10
  - platform: cbm7320
    name: Left Slider E
    channel: 11
  - platform: cbm7320
    name: Left Slider F
    channel: 12
  - platform: cbm7320
    name: Left Slider G
    channel: 13
  - platform: cbm7320
    name: Right Slider A
    channel: 0
  - platform: cbm7320
    name: Right Slider B
    channel: 1
  - platform: cbm7320
    name: Right Slider C
    channel: 2
  - platform: cbm7320
    name: Right Slider D
    channel: 3
  - platform: cbm7320
    name: Right Slider E
    channel: 4
  - platform: cbm7320
    name: Right Slider F
    channel: 5
  - platform: cbm7320
    name: Right Slider G
    channel: 6

sensor:
  - platform: da217
    address: 0x27
    resolution: 14bits
    full_scale: 4g
    tap_quiet_duration: 30ms
    tap_shock_duration: 70ms
    double_tap_duration: 500ms
    interrupt_source: unfiltered
    enable_double_tap_interrupt: true
    map_double_tap_interrupt_to_int1: true
    tap_acceleration_threshold: 0.0625
    output_data_rate: 125Hz
    accel_x:
      name: "Acceleration X"
    accel_y:
      name: "Acceleration Y"
    accel_z:
      name: "Acceleration z"
```

rootnegativ1, 2024-02-11:
@Ecco Thank you so much! 

FYI there's a key mismatch in your 6187 between ```OUTPUT_DATA_RATES``` in sensor.py and ```OutputDataRate``` in da217.h; one uses underscores before HZ but not the other.

happenpappen, 2024-06-16:
Sorry to hijack this issue for a slightly off-topic question, but: Could anyone provide a link to a guide on how to flash this light via the serial port? I have connected Vcc, GND, RX and TX to a serial ftdi adapter, but esptool is unable to read/write anything - so any help would be appreciated!

code-explorer1234, 2026-10-20:
Here's my working implementation by reverse engineering the original firmware dump.

```yaml
# ESPHome config for the Gosund LB3 RGBW lamp
#
# Hardware (from the stock Tuya firmware "cuco_bm02_xb_common" v1.0.1):
#   ESP8266EX, 2MB flash
#   GPIO15 red, GPIO12 green, GPIO5 blue, GPIO13 white (PWM)
#   I2C on GPIO2 (SDA) / GPIO14 (SCL), bit-banged:
#     0x2D  Chipsbank CBM7320 touch controller (15 pads)
#     0x27  MiraMEMS DA217 accelerometer, tap interrupt on GPIO4
#
# Touch pads: 7 colour pads, 7 brightness pads and a function pad.
#   Colour pad      colour mode: pick colour. White mode: acts as a brightness pad.
#   Brightness pad  7 levels, brightest first.
#   Function pad    short press toggles white/colour. A 6s hold fires an event only.
#   Double tap      on the lamp toggles power.
# Every pad and tap is also sent to Home Assistant as a "Touch" event.
#
# Required secrets: wifi_ssid, wifi_password, api_key, ota_password, ap_password
# First flash needs serial (3.3V USB-UART on TX/RX/GND, GPIO0 low at boot).
# Back up the stock firmware first: esptool read-flash 0 ALL backup.bin

substitutions:
  name: gosund-lb3
  friendly_name: Gosund LB3

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  on_boot:
    # Configure the accelerometer exactly as the stock firmware does, so a
    # double tap pulses INT1 (GPIO4).
    priority: 500
    then:
      - lambda: |-
          const uint8_t addr = 0x27;
          auto rd = [&](uint8_t reg, uint8_t &val) -> bool {
            if (id(bus_a).write(addr, &reg, 1) != i2c::ERROR_OK) return false;
            return id(bus_a).read(addr, &val, 1) == i2c::ERROR_OK;
          };
          auto wr = [&](uint8_t reg, uint8_t val) {
            uint8_t buf[2] = {reg, val};
            id(bus_a).write(addr, buf, 2);
          };
          uint8_t chip = 0;
          for (int i = 0; i < 6; i++) {
            if (rd(0x01, chip) && chip == 0x13) break;
            delay(20);
          }
          if (chip != 0x13) {
            ESP_LOGW("accel", "No accelerometer at 0x27 (chip id 0x%02X), tap disabled", chip);
            return;
          }
          uint8_t v = 0;
          if (rd(0x00, v)) wr(0x00, v | 0x24);  // soft reset
          delay(50);
          wr(0x0F, 0x01);  // range +-4g
          wr(0x11, 0x14);  // mode/bandwidth
          wr(0x10, 0x07);  // ODR, all axes
          wr(0x21, 0xEE);  // reset latch, ~50ms temporary latch
          wr(0x2A, 0x45);  // tap timing (double-tap window)
          wr(0x2B, 0x04);  // tap threshold, raise to make it less sensitive
          wr(0x19, 0x10);  // map double tap to INT1
          wr(0x16, 0x50);  // enable double tap (+ orient)
          wr(0x11, 0x04);  // normal mode
          ESP_LOGI("accel", "Accelerometer configured, tap interrupt on GPIO4");

esp8266:
  board: esp_wroom_02  # 2MB flash
[... 234 lines cut ...]
            } else if (key >= 1 && key <= 7) {
              // Colour pad: picks a colour in colour mode. In white mode the
              // stock firmware treats these pads as brightness pads too.
              int n = key;
              id(touch_event).trigger("colour_" + to_string(n));
              if (on) {
                if (is_white) {
                  colour_v = n;
                  apply(true, WHITE_LEVELS[n - 1] / 255.0f);
                } else {
                  colour_row = n;
                  colour_v = colour_col;
                  apply(false, COLOUR_V[colour_col - 1] / 255.0f);
                }
              }
            }
          }

          // Stock resets Wi-Fi after a ~6s hold. Here it only fires an event,
          // so you can attach your own action in Home Assistant.
          if (held == 20 && !fn_long_sent && millis() - fn_down_ms > 6000) {
            fn_long_sent = true;
            id(touch_event).trigger("function_long");
          }
```

pepsonEL, 2026-11-05:
> 21\. Here's my working implementation by reverse engineering the original firmware dump.
>      # ESPHome config for the Gosund LB3 RGBW lamp
>      #
>      # Hardware (from the stock Tuya firmware "cuco_bm02_xb_common" v1.0.1):
>      #   ESP8266EX, 2MB flash
>      #   GPIO15 red, GPIO12 green, GPIO5 blue, GPIO13 white (PWM)
>      #   I2C on GPIO2 (SDA) / GPIO14 (SCL), bit-banged:
>      #     0x2D  Chipsbank CBM7320 touch controller (15 pads)
>      #     0x27  MiraMEMS DA217 accelerometer, tap interrupt on GPIO4
>      #
>      # Touch pads: 7 colour pads, 7 brightness pads and a function pad.
>      #   Colour pad      colour mode: pick colour. White mode: acts as a brightness pad.
>      #   Brightness pad  7 levels, brightest first.
>      #   Function pad    short press toggles white/colour. A 6s hold fires an event only.
>      #   Double tap      on the lamp toggles power.
>      # Every pad and tap is also sent to Home Assistant as a "Touch" event.
>      #
>      # Required secrets: wifi_ssid, wifi_password, api_key, ota_password, ap_password
>      # First flash needs serial (3.3V USB-UART on TX/RX/GND, GPIO0 low at boot).
>      # Back up the stock firmware first: esptool read-flash 0 ALL backup.bin
>      
>      substitutions:
>        name: gosund-lb3
>        friendly_name: Gosund LB3
>      
>      esphome:
>        name: ${name}
>        friendly_name: ${friendly_name}
>        on_boot:
>          # Configure the accelerometer exactly as the stock firmware does, so a
>          # double tap pulses INT1 (GPIO4).
>          priority: 500
>          then:
>            - lambda: |-
>                const uint8_t addr = 0x27;
>                auto rd = [&](uint8_t reg, uint8_t &val) -> bool {
>                  if (id(bus_a).write(addr, &reg, 1) != i2c::ERROR_OK) return false;
>                  return id(bus_a).read(addr, &val, 1) == i2c::ERROR_OK;
>                };
>                auto wr = [&](uint8_t reg, uint8_t val) {
>                  uint8_t buf[2] = {reg, val};
>                  id(bus_a).write(addr, buf, 2);
>                };
>                uint8_t chip = 0;
>                for (int i = 0; i < 6; i++) {
>                  if (rd(0x01, chip) && chip == 0x13) break;
>                  delay(20);
>                }
>                if (chip != 0x13) {
>                  ESP_LOGW("accel", "No accelerometer at 0x27 (chip id 0x%02X), tap disabled", chip);
>                  return;
>                }
>                uint8_t v = 0;
>                if (rd(0x00, v)) wr(0x00, v | 0x24);  // soft reset
>                delay(50);
>                wr(0x0F, 0x01);  // range +-4g
>                wr(0x11, 0x14);  // mode/bandwidth
>                wr(0x10, 0x07);  // ODR, all axes
>                wr(0x21, 0xEE);  // reset latch, ~50ms temporary latch
>                wr(0x2A, 0x45);  // tap timing (double-tap window)
>                wr(0x2B, 0x04);  // tap threshold, raise to make it less sensitive
>                wr(0x19, 0x10);  // map double tap to INT1
>                wr(0x16, 0x50);  // enable double tap (+ orient)
>                wr(0x11, 0x04);  // normal mode
>                ESP_LOGI("accel", "Accelerometer configured, tap interrupt on GPIO4");
>      
>      esp8266:
>        board: esp_wroom_02  # 2MB flash
>        restore_from_flash: true
>      
[... 233 lines cut ...]
>                    // Colour pad: picks a colour in colour mode. In white mode the
>                    // stock firmware treats these pads as brightness pads too.
>                    int n = key;
>                    id(touch_event).trigger("colour_" + to_string(n));
>                    if (on) {
>                      if (is_white) {
>                        colour_v = n;
>                        apply(true, WHITE_LEVELS[n - 1] / 255.0f);
>                      } else {
>                        colour_row = n;
>                        colour_v = colour_col;
>                        apply(false, COLOUR_V[colour_col - 1] / 255.0f);
>                      }
>                    }
>                  }
>                }
>      
>                // Stock resets Wi-Fi after a ~6s hold. Here it only fires an event,
>                // so you can attach your own action in Home Assistant.
>                if (held == 20 && !fn_long_sent && millis() - fn_down_ms > 6000) {
>                  fn_long_sent = true;
>                  id(touch_event).trigger("function_long");
>                }

Your config works ok ?

code-explorer1234, 2026-11-05:
Yes. I have it on three devices - works perfectly