esphome/feature-requests #1097: Signal strength (proxy) for xiaomi_lywsd03mmc

samuelalexmclean, 2021-01-24:
**Describe the problem you have/What new integration you would like**

I have 4 xiaomi_lywsd03mmc temp / humidity sensors scattered around my house. I currently optimize their locations by moving my Node MCU (the BLE master) or the sensors (BLE slaves) to slightly different positions, then wait for ~12 hours of data to come through. I can tell a good placement by nice consistent readings. A bad placement drops a couple; even truly awful placements will get a few readings in over a 12 hour period.

It would be great if the ESP could track and report (as a first class metric, or in some diagnostic fashion) the average number of reads per hour per device, so I could see if some are doing better than others a little easier. This would also work quite well as a chart to notice when one of them goes offline, since the battery reporting doesn't work (I believe this is a hardware issue, nothing to do with ESPHome).

**Please describe your use case for this integration and alternatives you've tried:**

This would be helpful for both device placement / optimization, and dead device detection.

**Additional context**

You can do this manually but it's a lot of work looking at charts.

ejalal, 2021-02-11:
I had similar problem where my upstairs sensors were rarely reachable from the esp32 I have downstairs. I ended up buying a second esp32 for upstairs and configured both esp32 boards in a way to report the readings of all the sensors on the same mqtt topic. Should definitely fix your problem and prevent the hassle of placing them all the time.

nagyrobi, 2022-07-07:
A few tips for optimizing reception: https://esphome.io/components/sensor/xiaomi_ble.html?highlight=lywsd03mmc#improving-reception-performance