esphome/feature-requests #1099: Support for 1-wire  DS2408  binary_sensor and switch

brano1990, 2021-01-24:
**Describe the problem you have/What new integration you would like**

I would like to use communication between 1-wire binary_sensor and switch using DS2408 8-ch or DS2406 2-ch. It can be used in all directions, such as Relay control, status LED control via switch and binary sensor.
https://datasheets.maximintegrated.com/en/ds/DS2408.pdf

**Please describe your use case for this integration and alternatives you've tried:**

The problem is that I only have 3-wire attached to the buttons on the wall. I tried to use ADC 1115 as recommended, but this solution only works when using 3 - 6 buttons (I have 13 of them), above 6 I already have a problem using the "on_multi_click" function and I also have a problem with ADC when using ttp223 touch sensor . With 13 buttons, the voltage difference cannot be perfectly distinguished (fluctuates when the touch sensor is pressed). DALLAS ds2408 would solve this problem.
The ADC can be used with analog buttons. When using the ttp223 touch sensor, there is a high voltage fluctuation, especially for "on_multi_click". This is not very applicable to "on_multi_click", because only 6 buttons worked on 100% from using 13 buttons. If the switch could also be integrated into Dallas, I would be able to send 1-wire status using an LED. When using Dallas, it is possible to push 2 buttons at the same time, which is not possible with ADC.

**Additional context**

The integration of this solution into ESPhome would facilitate work over 1-wire with the used binary sensor and switch, which would probably be appreciated by several users who have few cables or it could also be used for the old 3-wire electrical installation.
[image]

randybb, 2021-01-24:
Basically related to https://github.com/esphome/feature-requests/issues/432

brano1990, 2021-01-25:
@randybb and came to a conclusion?
is any code working?
or Custom Component?

randybb, 2021-01-25:
I haven't noticed that there was an update from somebody, so I have no idea. Need to test it. I added it as a refference that more people want to have this integration +  we need to modify dallas component, as it supports temp sensors only.

JayElDubya, 2021-02-17:
@randybb I updated #432 with some success on the DS2413 (dual channel) onewire switch. As I noted in the post, it ONLY supports switch mode at the moment but as inefficient as it is, I think the code could be modified to work for the DS2408 as well.

-J

brano1990, 2021-02-17:
@JayElDubya  hi it only supports one device?

JayElDubya, 2021-02-17:
Only one DS2413 device (which can switch two devices).