esphome/feature-requests #1101: Add battery level sensor in LYWSD02 platform

Arman-GeDon, 2021-01-27:
**Describe the problem you have/What new integration you would like**

I would like to have battery level sensor in LYWSD02 platform for Xiaomi Mijia BLE Sensors

**Please describe your use case for this integration and alternatives you've tried:**

Currently LYWSD02 platform does not support battery level sensor.
But starting from firmware 1.1.2_00085 and later this device advertise its battery level.
It's supported already in other integration like Passive Ble Monitor:
https://github.com/custom-components/ble_monitor

And other software like Mijia Temp for Android.
So it is possible.

**Additional context**

fabiosoft, 2021-01-29:
I made some invastigation on this because i'am interested too... i noted that last time i run out battery not warning at all and display stuck on time (thanks e-ink)

The operating range is roughly between 2.1v to 3v.
Also, the Service: `ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6` with charateristics: `ebe0ccc4-7a0a-4b0c-8a1a-6ff2997da3a6` changes value from 0x64 to 0x0A (I guess it is 100% to 10%). This change is not immediately notified
I guess it's a binary indicator of battery status.

Now i have 0x24... so i hope we can assume a simple conversion in python but i'am not sure about.

```
>>> int("0x24", 16)
36
```

Let me know about your values

Arman-GeDon, 2021-02-21:
Hi. Thankfully from version 1.16 battery level is supported

fabiosoft, 2021-02-23:
Updated my tracker and works very well!
Thanks!

ChefkeGremmen, 2021-03-12:
Works great here as well, thx!

closed 2022-01-21 by rradar