esphome/feature-requests #1103: Replacement for TX20/TX23 Wind Speed/Direction Sensor

IkaSGC, 2021-01-30:
**Describe the problem you have/What new integration you would like**
I would like to add a wind and direction sensor to my weather station. Unfortunately, the TX20 / TX23 sensor is no longer manufactured and sold. There are several ways to replace the device. That is why I do not mention any specific ones.

**Please describe your use case for this integration and alternatives you've tried:**
weather stations

**Additional context**
I couldn't find any other used weather station anywhere.

nagyrobi, 2022-07-19:
Seems like WS2300-15 is something similar.
Also looks close to TX20 as: https://tasmota.github.io/docs/TX2x/

nagyrobi, 2024-05-19:
There's a wind speed meter on Aliexpress (black plastic), type **pr-3000-fsjt-n01** works with this config, using an RS485/TTL adapter (connected alone):
```yaml
uart:
  rx_pin: GPIOXX
  tx_pin: GPIOXX
  baud_rate: 4800

modbus:
  id: modbus1

modbus_controller:
- id: wind_speed_meter
  address: 1
  modbus_id: modbus1
  setup_priority: -10
  command_throttle: 200ms
  update_interval: 30s #or whatever you need

sensor:
- platform: modbus_controller
  modbus_controller_id: wind_speed_meter
  name: "Wind speed"
  device_class: wind_speed
  id: wind_speed
  register_type: read
  address: 1
  unit_of_measurement: "m/s"
  value_type: U_WORD
  accuracy_decimals: 1
  filters:
    - multiply: 0.1
```

They also sell a direction sensor, type **pr-3000-fxjt-n01** this config, using an RS485/TTL adapter (connected alone):

```yaml
uart:
  rx_pin: GPIOXX
  tx_pin: GPIOXX
  baud_rate: 4800

modbus:
  id: modbus1

modbus_controller:
- id: wind_direction_meter
  address: 1
  modbus_id: modbus1
  setup_priority: -10
  command_throttle: 200ms
  update_interval: 30s #or whatever you need

sensor:
- platform: modbus_controller
  modbus_controller_id: wind_direction_meter
  name: "Wind direction gear"
  icon: mdi:compass-rose
  register_type: read
  address: 1
  value_type: U_WORD
  accuracy_decimals: 0

- platform: modbus_controller
  modbus_controller_id: wind_direction_meter
  name: "Wind direction degree"
  icon: mdi:compass-rose
  register_type: read
  address: 2
  value_type: U_WORD
  accuracy_decimals: 0
  unit_of_measurement: "°"
```

Note that both **pr-3000-fsjt-n0**1 and **pr-3000-fxjt-n01** are by default set to modbus address `1`, so out of the box it's not possible to connect to the same modbus adapter.

nagyrobi, 2024-05-19:
The seller offers a helper application for Windows, called `485 Parameter Configuration Tool V5.0.7.1.exe`. The sensor can be connected to the PC with a USB-to-RS485 adapter, and the configuration tool makes it easily possible to change the modbus address to something else, eg. `2` (just type it in the _Addr_ box and press _Setup_ button).

After that it becomes possible to simply connect the sensors in parrallel on the same cable, to a single RS485-TTL adapter attached to a single UART on the ESP:

```yaml
uart:
  rx_pin: GPIOXX
  tx_pin: GPIOXX
  baud_rate: 4800

modbus:
  id: modbus1

modbus_controller:
- id: wind_speed_meter
  address: 1
  modbus_id: modbus1
  setup_priority: -10
  command_throttle: 200ms
  update_interval: 30s #or whatever you need

- id: wind_direction_meter
  address: 2
  modbus_id: modbus1
  setup_priority: -10
  command_throttle: 200ms
  update_interval: 30s #or whatever you need

sensor:
- platform: modbus_controller
  modbus_controller_id: wind_speed_meter
  name: Szélsebesség
  device_class: wind_speed
  id: wind_speed
  register_type: read
  address: 1
  unit_of_measurement: "m/s"
  value_type: U_WORD
  accuracy_decimals: 1
  filters:
    - multiply: 0.1

- platform: modbus_controller
  modbus_controller_id: wind_direction_meter
  name: "Wind direction gear"
  icon: mdi:compass-rose
  register_type: read
  address: 1
  value_type: U_WORD
  accuracy_decimals: 0

- platform: modbus_controller
  modbus_controller_id: wind_direction_meter
  name: "Wind direction degree"
  icon: mdi:compass-rose
  register_type: read
  address: 2
  value_type: U_WORD
  accuracy_decimals: 0
  unit_of_measurement: "°"
```

SeByDocKy, 2024-05-19:
I ordered this one also in modbus/RS485.... : https://fr.aliexpress.com/item/1005006174404179.html?spm=a2g0o.order_list.order_list_main.50.188a5e5b1rIDLR&gatewayAdapt=glo2fra

The precision seems better. I will write a native ESPhome conponent for ... even if it's 2 simple modbus sensors.

nagyrobi, 2024-05-25:
Done:
https://devices.esphome.io/devices/Renke-RS-FSJT-N01-Wind-Speed
https://devices.esphome.io/devices/Renke-RS-FXJT-N01-Wind-Direction

Both can be purchased on Aliexpress and similar.

closed 2024-05-25 by nagyrobi

gitsmarc, 2024-06-25:
> The seller offers a helper application for Windows, called `485 Parameter Configuration Tool V5.0.7.1.exe`. The sensor can be connected to the PC with a USB-to-RS485 adapter, and the configuration tool makes it easily possible to change the modbus address to something else, eg. `2` (just type it in the _Addr_ box and press _Setup_ button).
> 
> After that it becomes possible to simply connect the sensors in parrallel on the same cable, to a single RS485-TTL adapter attached to a single UART on the ESP (kept the default address `1` for the speed sensor and changed the direction sensor to `2`):
> 
> ```yaml
> uart:
>   rx_pin: GPIOXX
>   tx_pin: GPIOXX
>   baud_rate: 4800
> 
> modbus:
>   id: modbus1
> 
> modbus_controller:
> - id: wind_speed_meter
>   address: 1
>   modbus_id: modbus1
>   setup_priority: -10
>   command_throttle: 200ms
>   update_interval: 30s #or whatever you need
> 
> - id: wind_direction_meter
>   address: 2
>   modbus_id: modbus1
>   setup_priority: -10
>   command_throttle: 200ms
>   update_interval: 30s #or whatever you need
> 
> sensor:
> - platform: modbus_controller
>   modbus_controller_id: wind_speed_meter
>   name: "Wind Speed"
>   device_class: wind_speed
>   id: wind_speed
>   register_type: read
>   address: 0
>   unit_of_measurement: "m/s"
>   value_type: U_WORD
>   accuracy_decimals: 1
>   filters:
>     - multiply: 0.1
> 
> - platform: modbus_controller
>   modbus_controller_id: wind_direction_meter
>   name: "Wind direction gear"
>   internal: true
>   register_type: read
>   address: 0
>   value_type: U_WORD
>   accuracy_decimals: 0
>   on_value:
>     - lambda: |-
>           switch (int(x)) {
>             case 0:
>               id(wind_direction_cardinal).publish_state("N");  //North
>               break;
>             case 1:
>               id(wind_direction_cardinal).publish_state("NE"); //North-East
>               break;
>             case 2:
>               id(wind_direction_cardinal).publish_state("E");  //East
>               break;
>             case 3:
>               id(wind_direction_cardinal).publish_state("SE"); //South-East
>               break;
>             case 4:
>               id(wind_direction_cardinal).publish_state("S");  //South
>               break;
>             case 5:
>               id(wind_direction_cardinal).publish_state("SW"); //South-West
>               break;
>             case 6:
>               id(wind_direction_cardinal).publish_state("W");  //West
>               break;
>             case 7:
>               id(wind_direction_cardinal).publish_state("NW"); //North-West
>               break;
>             default:
>               id(wind_direction_cardinal).publish_state("");   //invalid
>           }
> 
> - platform: modbus_controller
>   modbus_controller_id: wind_direction_meter
>   name: "Wind direction degree"
>   icon: mdi:compass-rose
>   register_type: read
>   address: 1
>   value_type: U_WORD
>   accuracy_decimals: 0
>   unit_of_measurement: "°"
> 
> text_sensor:
>   - platform: template
>     name: "Wind direction cardinal"
>     id: wind_direction_cardinal
>     icon: mdi:compass-rose
> ```

Hi Just to inform you that the change of address can be done directly in home assistant without an usb to rs485 just add those line and then you will have access to change it ether in the main table or if you visit you esp32.  Note after you change the address the sensor will stop updating, you need to change the address in your code and reflash. dont forget to comment out the change of address since you only need it once.

uart:
  rx_pin: GPIOXX
  tx_pin: GPIOXX
  baud_rate: 4800

modbus:
  id: modbus1

modbus_controller:
- id: wind_direction_meter
  address: 1
  modbus_id: modbus1
  setup_priority: -10
  command_throttle: 200ms
  update_interval: 30s #or whatever you need

- platform: modbus_controller
  modbus_controller_id: wind_direction_meter
  name: "address"
  icon: mdi:compass-rose
  register_type: read
  address: 0x07D0
  value_type: U_WORD

number:
  - platform: modbus_controller
    modbus_controller_id: wind_direction_meter
    id: change_address
    name: "change_address"
    address: 0x07D0
    register_type: holding
    value_type: U_WORD

nagyrobi, 2024-06-26:
Added to the doc.