esphome/feature-requests #1107: Let the "partition" light combine single lights

HACKER-3000, 2021-02-01:
**Describe the problem you have/What new integration you would like**

The partition light should be able to combine single lights, so you can attach a few LEDs to your ESP and have them all as one Light that can still have adressable effects and such.

**Please describe your use case for this integration and alternatives you've tried:**

I connect 3 RGB LEDs and combine them into an 3 pixel/led ""strip""

**Additional context**

i hope tis is not too hard to implement

paulmonigatti, 2021-09-23:
This feature has just been merged into dev (esphome/esphome#2256).
This PR should also implement #734, #910, and #966

jesserockz, 2021-09-23:
Closed by esphome/esphome#2256

closed 2021-09-23 by jesserockz