esphome/feature-requests #1108: Implement Expressif ESP32 &  ESP8266 Node Wifi Mesh into ESPHome core

DayRunner131, 2021-02-01:
**Describe the problem you have/What new integration you would like**

Node Mesh has been implemented in Zigbee & Z-Wave systems which greatly increases the range and throughput of the network.

This request is to integrate Mesh Network topology into ESPHome nodes.

By scattering nodes across the home, one could create a blanket of BLE repeaters, or even reduce gaps in Wifi node coverage without the use of a separate Wifi AP.

**Please describe your use case for this integration and alternatives you've tried:**

Expressif has released mesh SDK's for the ESP32 & 8266 chipsets. This feature request is to incorporate Wifi mesh into ESPHome to allow greater node range & throughput.

See: https://www.espressif.com/en/products/sdks/esp-wifi-mesh/overview
&
https://randomnerdtutorials.com/esp-mesh-esp32-esp8266-painlessmesh/

**Additional context**

cwhzzang, 2021-03-04:
As a provider of wifi functionality, it is the only alternative to large-capacity local communication. I need your support

DayRunner131, 2021-03-04:
I'm reading through my original request and perhaps I missed some specifics. The request is for ESPHome nodes to mesh into their own "ESPHome" network which would then feed into the home Wifi network. I.e. if ESPHome node 8 has a weak/no Wifi signal from the home network but can speak to ESPHome node 4 strongly, it could relay the information through node 4 back to the home network. Zigbee and Zwave do this through "hops" from a node, to a relay, then to the controller when a node is far from the controller. You can specify at the controller level how many "hops" are allowed back to the controller. This is usually 1 to 2 hops. In my configuration, I have a ESPHome node in my garage that has a very weak wifi signal but would be able to communicate with a node closer to my home wifi antenna.

mdewyer, 2021-04-20:
Would love to see this. Subbing!

ackerschott, 2021-04-29:
great feature, that I also apreciate

ljohnso16, 2021-05-05:
how do i enable in my esphome yaml?

andyfsimon, 2021-05-06:
I am upvoting this. According to the docs above, it will require adding the painlessmesh library and changing the way the whole network is dealt with in esphome, but seems doable.

timohi, 2021-05-06:
I would be very happy about the feature.

andyfsimon, 2021-05-12:
Well, I did have a look at it, the problem I fear is that it would require a complete redesign of the network handling.
See, if implementing that library means handling the communication for the entire mesh **within each esphome node**. As of now wifi is pretty much given for granted and every esphome node uses it for it's own.

I do strongly believe that esphome should go down that path, but I am not blind to the fact that it would mean esphome 2.0.

roscoegray, 2021-08-03:
Would something like this need to be implemented to support the use and connection to BLE mesh server devices? Some devices I have the [BT5s Module](https://developer.tuya.com/en/docs/iot/bt5s-datasheet?id=K98sb3vijku2u) from Tuya, and advertise as a mesh beacon.

pupupupupup, 2021-11-13:
I would be very happy about the feature.

Hohenloherin, 2022-06-28:
I have no clue what it would mean, but I'd appreciate this feature very much!

jigassa, 2023-08-01:
UP!
I vote for it

shiwzhi, 2023-10-21:
It seems that [https://github.com/espressif/esp-mesh-lite](url) can provide each node with IP connectivity while also supporting mesh functionality.

hevnsnt, 2023-11-04:
It seems like [https://github.com/espressif/esp-mesh-lite](https://github.com/esphome/feature-requests/issues/url) (as mentioned above is a perfect solution.

rajtan, 2023-12-04:
I would also upvote this. I firmly believe that this would open up newer possibilities in moving forward with decentralized automations as well.

saya6k, 2023-12-17:
I upvote this.

FARKIr, 2024-07-21:
I have 70 ESP devices spread around my house. In some areas, I have a weak signal and need to use a mesh network with 5 pieces Tenda wifi . I think this is unnecessary and a waste of energy and money. Implementing WiFi mesh into ESPHome could solve many problems I have, such as some ESP devices constantly disconnecting from the network.I believe this functionality would greatly help in ensuring a stable connection for all ESP devices without the need for additional WiFi access points. I would greatly appreciate it if this feature could be added to ESPHome. Thank you.