esphome/feature-requests #1110: SIM800L support for voice calls and/or Flash SMS

remedix, 2021-02-02:
**Describe the problem you have/What new integration you would like**

Support for Voice Calls and/or Flash SMS for SIM800L - Currently only sending and receiving text is supported. However sometimes just calling might be a better options. Also a good idea is to have support for Flash SMS. In my case my carrier sends a Flash SMS with my balance and my contract expiration. It would be nice to capture that from esphome.

**Please describe your use case for this integration and alternatives you've tried:**

Did not try other alternatives

**Additional context**

douglasfranciscon, 2022-02-02:
Any chance of having a call_received or being able to collect the signal strength?

nagyrobi, 2022-07-06:
Duplicate of https://github.com/esphome/feature-requests/issues/1347

closed 2022-07-06 by nagyrobi