esphome/feature-requests #1111: ESP32 based Zigbee2mqtt gateway

koldogut, 2021-02-02:
**Describe the problem you have/What new integration you would like**

The idea is to include support for zigbee2mqtt in ESP32 boards using wired devices such as CC2530, CC2531, etc 

**Please describe your use case for this integration and alternatives you've tried:**

I'm working in a security solution for cars using a cheap coordinator (ESP32+CC2530) plus zigbee sensors (vibration, presence and open-close) but it can be used also as a **cheap gateway** to **extend the zigbee network** using the **wifi coverage** at home, in the garden, basements, etc.

There is a russian project to achieve [this](https://slsys.github.io/Gateway/web_eng.html) and works great, but I'll prefer to have it already included under my ESPHome ecosystem. Morever, support is in russian and despite users have gently helped me with the setup,  it's difficult to find solutions or explain issues using the translator.

**Additional context**

Here you are the link to the official suppor thread: https://slsys.github.io/Gateway/web_rus.html

The idea is just to **port this into ESPHome** and make those functionalities available joint together with the existing ones.

[image]

manju-rn, 2021-02-15:
This would be really great if this could act as a co-ordinator, I currently use the CC2531 in RPI, However, porting this in ESP would mean that it could be kept away from RPI and still use the zigbee2mqtt for a certain place where the range could be maximized

tjunussov, 2021-04-17:
As you know Tasmota already ported Zigbee https://tasmota.github.io/docs/Zigbee/ functionality

Hedda, 2021-11-12:
Duplicate of https://github.com/esphome/feature-requests/issues/681

FYI, there is also a somewhat related request posted here  but with Serial-to-IP server instead of MQTT so not really the same at all -> https://github.com/esphome/feature-requests/issues/688

PS: Please note that #681 and #688 are two sperate requests with different goals.

Gio-dot, 2021-11-12:
> **Describe the problem you have/What new integration you would like**
>
> The idea is to include support for zigbee2mqtt in ESP32 boards using wired devices such as CC2530, CC2531, etc
>
> **Please [...]

Take a look here: https://github.com/Gio-dot/Z-Bee-Duo

I produce a modular usb coordinator that can also work with wifi or Ethernet connectivity.

Hedda, 2021-11-16:
There is also ZigStar LAN Coordinator and Tube's Zigbee Gateways (CC2652P variants) which are both open-source hardware:

https://zig-star.com/projects/zigbee-gw-lan/

https://github.com/mercenaruss/zigstar_gateways

https://www.tubeszb.com

https://github.com/tube0013/tube_gateways/

Those have benefits of their makers being active members of the Home Assistant community and their support so far is great!

https://community.home-assistant.io/t/zigstar-zigbee-coordinators-and-routers/338586/

https://community.home-assistant.io/t/tubes-zb-coordinators-and-routers-was-zigbee-router-on-steroids/280896

nagyrobi, 2022-07-01:
dup of https://github.com/esphome/feature-requests/issues/681

closed 2022-07-01 by nagyrobi