esphome/feature-requests #1112: Add support for HA attributes

mikosoft83, 2021-02-03:
**Describe the problem you have/What new integration you would like**
Instead of using template sensors to extraxt attributes, add ability to read attributes directly in ESPHome.

**Please describe your use case for this integration and alternatives you've tried:**
It makes sharing reusable code (like my own flexible menu system) much easier, since there is no need to handle two configs (ESPH and HA).

**Additional context**

ioios-io, 2021-02-03:
Seconded. The ability to share ESPHome software without having to add template sensors into the user's HA config makes sharing our ESPHome-based software so much easier for casual users. I'm praying for this one. Thanks devs for all you do.

IainPHay, 2021-02-04:
If possible this would be great. ESPHone is absolutely wonderful but my skills are not up to the level yet where I can use this kind of described project without a great deal of head scratching!

frenck, 2021-05-11:
Last night I've been playing a bit with this:

[image]

ESPHome configuration is visible in the top right.

This change/addition does require changes to Home Assistant itself as well. So this will take some time in order for everything to update and process the changes.

I'll start opening up PRs for code reviewing slowly.

rradar, 2021-06-19:
@mikosoft83 support for attributes is now present

closed 2021-06-19 by rradar