esphome/feature-requests #1113: H-Bridge Light Color Tone/Temperature

OvCa77, 2021-02-03:
**Describe the problem you have/What new integration you would like**

H-Bridge Light Color Tone/Temperature
**Please describe your use case for this integration and alternatives you've tried:**

I would like to propose new integration/modification of existing module, which will allow to change color temperature, for light such as Arilux ceiling light, which uses two channels, one to control brightness, other to set temperature of white light. https://esphome.io/components/light/hbridge.html
**Additional context**

jesserockz, 2021-02-16:
https://github.com/ssieb/custom_components/tree/master/cwww2

nagyrobi, 2022-07-07:
Implemented https://esphome.io/components/light/cwww.html

closed 2022-07-07 by nagyrobi