esphome/feature-requests #1117: expect ModbusDevice support specified uart_id

bestlong, 2021-02-05:
**Describe the problem you have/What new integration you would like**

when multiple uart, ModbusDevice config can manually specify uart_id.

**Please describe your use case for this integration and alternatives you've tried:**

use ESP32 and two or three PZEM004Tv3.0 device.

PZEM004Tv3.0 use TTL RS232, Modbus-RTU

pzem004t component not support PZEM004Tv3.0 device, use pzemac component is work. But only one device can be connected.

**Additional context**

Gio-dot, 2021-02-05:
> **Describe the problem you have/What new integration you would like**
>
> when multiple uart, ModbusDevice config can manually specify uart_id.
>
> **Please describe your use case for this integration and [...]

It is already supported. Here an example: 
https://community.home-assistant.io/t/how-to-use-pzem004t-energy-monitor-with-esphome/107083/54?u=gt4020

bestlong, 2021-02-05:
Thank @Gio-dot 

https://community.home-assistant.io/t/how-to-use-pzem004t-energy-monitor-with-esphome/107083/106

I see this part

```
modbus:
  id: mbus
  uart_id: ubus
  
...
sensor:
  - platform: pzemac
    address: 1
    modbus_id: mbus
...
```

closed 2021-02-05 by bestlong