esphome/feature-requests #1118: Loop current song for Rtttl Buzzer?

whc2001, 2021-02-05:
**Describe the problem you have/What new integration you would like**
I am trying to hack a siren, so I would like to play the song sent through MQTT repeatedly until `rtttl.stop` is invoked.

**Please describe your use case for this integration and alternatives you've tried:**
Not aware there are any other options for now.

**Additional context**

glmnet, 2021-02-06:
You can do that with node automations 

If not sure how to ask in discord they will get you going very quickly

whc2001, 2021-02-06:
> You can do that with node automations
> 
> If not sure how to, ask in discord they will get you going very quickly

Thanks, do you mean using NodeRED to achieve this? But I wanted it to be self enclosed without relying on other softwares. I have found a solution with `script` though:

```
globals:
  - id: score
    type: std::string

# play_song() { while(1) { rtttl.play(score); while(rtttl.is_playing); } }
script:
  - id: play_song
    then:
      - while:
          condition:
            - lambda: 'return true;'
          then:
            - rtttl.play:
                rtttl: !lambda 'return id(score);'
            - wait_until:
                - not:
                    rtttl.is_playing

mqtt:
  on_message:
    - topic: play
      then:
        - lambda: id(score) = x;
        - script.execute: play_song

    - topic: stop
      then:
        - script.stop: play_song
        - rtttl.stop
        - lambda: id(score) = "";
```

closed 2021-02-06 by whc2001

glmnet, 2021-02-06:
No. I meant something like you did 

You could have also used `on_finished_playback` trigger of the rtttl. 

But if it's working already it's ok!

whc2001, 2021-02-06:
> No. I meant something like you did
> 
> You could have also used `on_finished_playback` trigger of the rtttl.
> 
> But if it's working already it's ok!

I tried execute `rtttl.play` again inside `on_finished_playback` but it only output the "Playing XX" message without actual producing sound and I don't know why. The infinite loop script is kinda like a hack but at least it works.

glmnet, 2021-02-06:
Ok I'll try it myself later. There might be a bug