esphome/feature-requests #1119: People counter using VL53L1X sensor

Andrea-Fox, 2021-02-05:
**Describe the problem you have/What new integration you would like**

A people counter which uses the VL53L1X sensor in order to understand if someone is getting in or out from a room

**Please describe your use case for this integration and alternatives you've tried:**

This sensor is able to measure the distance from another point and if positioned above a door, it can detect someone passing under it. Differently from the vl530x (already integrated in esphome), this sensor is also able to detect in which direction the person is moving, hence it can understand if someone is getting in or out of the room. If placed in the correct place, it can be used to count people in the room with great precision.

**Additional context**

I already wrote some esp32/esp8266 code to cope with this issue, which can be found [here](https://github.com/Andrea-Fox/peopleCounter), based on the Adafruit vl53l1x library. Unfortunately I don't have the abilities to integrate the code here, but someone else might be able to do so.

rradar, 2021-02-08:
https://github.com/DutchDeffy/esphome_custom_component_VL53L1X_people_counter

jesserockz, 2021-02-16:
esphome/esphome#1539 is a start

Lyr3x, 2021-08-20:
I initially started the project 3 years ago and revised it in the last days:
https://github.com/Lyr3x/Roode
The implementation does utilise ESPHome now and uses ST's algorithm to count people.

closed 2022-07-06 by nagyrobi