esphome/feature-requests #1120: Support for MPU6886 6 axis inertial sensor

lettore, 2021-02-06:
**Describe the problem you have/What new integration you would like**

Add native support for the inertial 6 axis sensor MPU6886.
**Please describe your use case for this integration and alternatives you've tried:**

I'm using boards from M5Stack like the Atom Matrix https://docs.m5stack.com/#/en/core/atom_matrix equipped with ESP32 pico, this boards come with a MPU6886 inertial sensor connected to the I2C bus. I think it's a nice board for this project as it comes already packaged and with expansion modules.
I tried configuring it as the MPU6050, the I2C scanner find one device at 0x68 as it should but at the end it's unable to talk with the sensor, obviously because is a different one.
Looking on the web I found very little info about this sensor, as it seems is used only by M5Stack, there is a library for Arduino https://www.arduino.cc/reference/en/libraries/i2c-mpu6886-imu/  but I'm not really good enough to use that to make it functioning on ESPHome.
I hope this will be useful also for the community.

**Additional context**

richardklingler, 2021-03-28:
It is almost the same one....

Change whoami id from 0x68 to 0x19 in the setup function...

hmax42, 2021-12-18:
duplicating the esphome component mpu6050 and replacing every occurrence of 6050 with 6886 (with sed) and chaing the who_am_i_expected result in line 26 to 0x19 still yields a failing component

ts, 2022-01-23:
I'd very much like to get the motion data from the handful of m5 Atom devices I have in ESPHome. Pretty new to microcontrollers but I'm happy to dive in if someone can point me in the right direction.

hmax42, 2022-01-24:
@ts the mpu6050 lib of esphome is here https://github.com/esphome/esphome/tree/dev/esphome/components/mpu6050

esphome did use the code i created, saw it in the compile log, but the runtime log then said, it doesn't work

fabaff, 2022-02-09:
It would be appreciated if https://github.com/esphome/esphome/pull/3183 would be tested on different hardware. It works for me on the M5Stick C, the Atom and with the 6-Axis IMU Unit.

I don't have access to a M5StickC Plus or a M5Stack Core2.

Thanks

closed 2022-02-20 by jesserockz