esphome/feature-requests #1122: Platform cse7766: use pulses for total energy calculation

micw, 2021-02-06:
**Describe the problem you have/What new integration you would like**

The cse7766 platform currently reads voltage, current and power from the sensor. To calculate the energy consumption over a time period, one must integrate over the power which requires a high time resolution for precise values. This is not possible with this implementation.

The cse7766 itself provides a pulse counter which gives a wat-per-time information that is suitable to calculate the energy consumtion over time (e.g. how much kilowatt-hours have been consumed in a given time). This should be supported by the platform

**Please describe your use case for this integration and alternatives you've tried:**

Documentation suggest to use total_daily_energy platform. Drawback is that it works only for very constant loads. For changing loads it's very in-accurate.

**Additional context**

The specification can be found at http://dl.itead.cc/S31/CSE7766.pdf (page 21 and following)

An implementation can be found in https://github.com/arendst/Tasmota/blob/ac054ea666a2e2ea764346db3b69bb1723df00f0/tasmota/xnrg_02_cse7766.ino

DTTerastar, 2021-08-05:
This would be nice to have with all the energy stuff in the new home assistant!

redpizza69, 2021-09-29:
+1 for this change

closed 2021-11-30 by oxan

micw, 2022-01-18:
@ianchi This is absolutely great. Nearly missed the feature since I did not see the mail stating it's closed. Thank you very much!

micw, 2022-01-20:
@ianchi, one more question. How does it work with calibration? I have found https://frenck.dev/calibrating-an-esphome-flashed-power-plug/ which uses a sensor filter to calibrate. But I assume that it's not honored in the code that multiples the pulses with the current.

I also have seen that a calibrated value is read from sensor data:
https://github.com/ianchi/esphome/blob/09c28ab36c2eda49eed1f261b6386da7494e9239/esphome/components/cse7766/cse7766.cpp#L85-L90

Do you know how that calibration actually works?

ianchi, 2022-02-04:
cse7766 has calibration data written in by the manufacturer. As it is a separate chip this data is retained after flashing ESPhome, so it should just work.

As this component has no parameter to manually adjust values, if outputs are not correct in your case, you can apply a filter to individual sensors to recalibrate them. But as you said, all sensors should be calibrated independently (voltage/current/power/energy).
In the case of the `energy` sensor you should apply a consistent calibration to the `power` one.
I think the easiest way would be to use the same `multiply` filter in both sensors. (you need to replace the `calibrate_linear` for the implicit factor, assuming you don't introduce an offset at zero)