esphome/feature-requests #1123: Platform cse7766: support for power factor

micw, 2021-02-06:
**Describe the problem you have/What new integration you would like**

It would be nice if the platform would provide "power factor" as quotient of active power and apparant power. It can easily be calculated outside the platform but I think it would fit very well in a power measurement platform.

**Please describe your use case for this integration and alternatives you've tried:**

For the power factor without the need of scripting the calculation outside of the platform.

**Additional context**

An imlementation can be found here: https://github.com/arendst/Tasmota/blob/d4ddc78e6289b4d45e4f0dbb47a40489bd14f768/tasmota/xdrv_03_energy.ino#L943

Markourai, 2021-11-03:
Yes, please support this platform (for my sonoff dual r3). 
It’s already the case on ESPEasy and Tasmota.
Thank you

Boh1, 2021-12-24:
I'm also interested in Power Factor support for the S31.

spikeygg, 2022-01-02:
Ditto on this request, very interested in native PF reporting on the Sonoff S31 in ESPHome.

mkaatman, 2022-01-16:
Hey Guys, Was having a discussion about this on discord and here's what we came up with: (plus some extras I had in my config)

```
sensor:
  - platform: cse7766 #https://esphome.io/components/sensor/cse7766.html
    power:
      name: $friendly_name Power
      unit_of_measurement: W
      id: wattage
      on_value:   # set or clear in_use template binary sensor depending on whether power usage is over threshold
        - if:
            condition:
              lambda: 'return (x >= id(threshold).state);'
            then:
              - binary_sensor.template.publish:
                  id: in_use
                  state: ON
            else:
              - binary_sensor.template.publish:
                  id: in_use
                  state: OFF
    current:
      id: current
      name: $friendly_name Current
      unit_of_measurement: A
    voltage:
      id: voltage
      name: $friendly_name Voltage
      unit_of_measurement: V
    update_interval: 10s # 20 second effective update rate for Power, 40 second for Current and Voltage.

  # Reports the total Power so-far each day, resets at midnight
  # See https://esphome.io/components/sensor/total_daily_energy.html
  - platform: total_daily_energy
    name: $friendly_name Total Daily Energy
    power_id: wattage
    filters:
        - multiply: 0.001  ## convert Wh to kWh
    unit_of_measurement: kWh

  - platform: template
    name: $friendly_name Power Factor
    id: power_factor
    lambda: return id(wattage).state / id(voltage).state / id(current).state;

number:      # used as a threshold for whether the plugged-in devices is running.
  - platform: template
    name: ${friendly_name} Use Threshold
    min_value: 1
    max_value: 100
    step: 1
    initial_value: 3
    id: threshold
    entity_category: config
    optimistic: true     # required for changing value from home assistant
    restore_value: true
    unit_of_measurement: Watt(s)
    mode: box
    on_value:
      - if:               # set or clear in_use template binary sensor depending on whether power usage is above threshold
          condition:
            lambda: 'return (id(wattage).state >= x);'
          then:
            - binary_sensor.template.publish:
                id: in_use
                state: ON
          else:
            - binary_sensor.template.publish:
                id: in_use
                state: OFF
```

From my individual device config I do something like:

```
substitutions:
  name: aquarium-bubbler
  friendly_name: Large Aquarium Bubbler
  device_description: Sonoff s31

packages:
    wifi: !include wifi.yaml
    s31: !include s31.yaml
    s31power: !include s31power.yaml ## I separate out power because I have some s31 lites too

esphome:
  name: $name
  comment: $device_description

# Enable Home Assistant API
api:

debug:
```