esphome/feature-requests #1125: 3.5" TFT with ILI9488

sj3fk3, 2021-02-08:
**Describe the problem you have/What new integration you would like**

Would love to see support for ILI9488 display driver to use this 3.5" TFT Display. https://www.makerfabs.com/esp32-3.5-inch-tft-touch-capacitive-with-camera.html

**Please describe your use case for this integration and alternatives you've tried:**

It's a reasonably priced devices with a large screen and cam. Good for access-control or HA control display. 

**Additional context**

Arduino drivers:
https://github.com/Makerfabs/Project_Touch-Screen-Camera

rlust, 2021-07-22:
Has anyone got this to work on ESPHome?

kuradi, 2021-12-09:
would love to see support for ILI9488

Thomasdmm, 2021-12-28:
One more for ili9488 support with esphome Thanks

Drumdevil, 2022-01-17:
And another for ILI9488 support!

sbur83, 2022-01-22:
I have the 3.95" mega shield that uses this driver too, it should also work with ESP32 tho. Would be awesome to get it into home assistant, even better if it could be done taking advantage of an ATmega2560 with built in ESP8266.

Zoogara, 2022-03-02:
This may be a trivial exercise - in this post https://www.instructables.com/Arduino-and-35-Inch-320x480-TFT-LCD-ILI9488-SPI-In/ it was stated that it was made to work with ILI9341 drivers by simply changing the height in the defines:

#define ILI9341_TFTWIDTH 320
#define ILI9341_TFTHEIGHT 480

And a change in the init:

writecommand(ILI9341_DFUNCTR); // Display Function Control
writedata(0x02);
writedata(0x02);
writedata(0x3B);

Maybe someone who reads this would be smart enough to create a new dev repo of ESPHome that we can modify and test - I know I am not - I have tried and failed...

Pseudomizer, 2023-01-28:
Hi guys, any update on this? I see it was added but not committed. I would also love to see support for Ili9488 and touch in ESPhome. https://www.makerfabs.com/esp32-s3-parallel-tft-with-touch-ili9488.html

nielsnl68, 2023-01-28:
I'm working on a new display component that uses Arduino_GFX as backbone. You can find it on my github at github.com/nielsnl68/esphome-components

jens-markussen, 2023-02-05:
Eagerly waiting for news on this one. Pls pls.

nielsnl68, 2023-02-05:
sorry, it takes me more time then i hopped.

GitMouth, 2023-02-21:
Hi ! Here to follow the ili9488 support !
Thank a lot for your job

nielsnl68, 2023-02-22:
I have ordered an ili9488 display recently hoping to find why that one is not working within esphome. I hope i will receive it within  Aa week or two.  So stay toned.

pascal-parent, 2023-05-04:
I am also having the problem, anyone found a solution?
https://www.makerfabs.com/esp32-s3-parallel-tft-with-touch-ili9488.html
I was hoping to use it to pull energy information from HA

GeraldPape, 2023-05-30:
any updates on this topic?