esphome/feature-requests #1126: UART hardware flow control

awasaw, 2021-02-08:
**Describe the problem you have/What new integration you would like**
Some devices require flow control when communicating via UART. In this case, it will be useful to add the ability to configure flow control in the UART component. ESP8266 has a hardware implementation of flow control(RTS, CTS, DTR, DSR), which is enabled through the UARTCONF register.
This improvement will make it possible to use MAX485 for communication under the RS485 protocol.

**Please describe your use case for this integration and alternatives you've tried:**
I am integrating the electricity meter through the TTL2RS485 converter based on the MAX485 chip, and I need to control the direction of information transfer. At first I wanted to implement a software switch, but delving deeper into the topic, I realized that it was enough to use the hardware capabilities of the board. 

**Additional context**
[ESP8266_NONOS_SDK/.../uart.h](https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/driver_lib/include/driver/uart.h)
```
typedef enum {
    USART_HardwareFlowControl_None    = 0x0,
    USART_HardwareFlowControl_RTS     = 0x1,
    USART_HardwareFlowControl_CTS     = 0x2,
    USART_HardwareFlowControl_CTS_RTS = 0x3
} UART_HwFlowCtrl;
```
[future-request: RS485 serial support / tx enable pin control]( https://github.com/esphome/feature-requests/issues/421) that can be resolved by this.

Gio-dot, 2021-02-08:
Ok for the request, but you can use MAX13487 instead (automatic flow control) or an adapter similar to: 
https://www.amazon.it/DSD-TECH-SH-U12-MAX13487-Raspberry/dp/B07B667STP/ref=sr_1_1?__mk_it_IT=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=max13487&qid=1612797862&sr=8-1
or:
https://www.amazon.it/Serial-Mutual-Conversion-Automatic-Control/dp/B07RKY1G71/ref=sr_1_2?__mk_it_IT=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=max13487&qid=1612797948&sr=8-2

awasaw, 2021-02-08:
> Ok for the request, but you can use MAX13487 instead (automatic flow control) or an adapter similar to (all with automatic flow control feature) :
In the case of the MAX485, it is possible to change the converter, but the next project will be a control system for the Chinese diesel air-heater "Hcalory" (clone Eberspächer), which uses UART on one wire(not 1-Wire). And I have a choice to use together NPN and PNP transistors to switch rx/tx as it was done in the original board, automatically switching them with RTS signal, or write a software implementation of UART over one wire. 
In any case, the ability to configure hardware flow control will be useful in the future.

michelebergo, 2021-03-28:
> Ok for the request, but you can use MAX13487 instead (automatic flow control) or an adapter similar to (all with automatic flow control feature) :
> https://www.amazon.it/DSD-TECH-SH-U12-MAX13487-Raspberry/dp/B07B667STP/ref=sr_1_1?__mk_it_IT=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=max13487&qid=1612797862&sr=8-1
> or:
> https://www.amazon.it/Serial-Mutual-Conversion-Automatic-Control/dp/B07RKY1G71/ref=sr_1_2?__mk_it_IT=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=max13487&qid=1612797948&sr=8-2
> or
> https://it.aliexpress.com/item/32864362561.html?spm=a2g0o.productlist.0.0.5503714fI17cfY&algo_pvid=0046253e-bee0-4b79-b8c4-71c4a32c57fd&algo_expid=0046253e-bee0-4b79-b8c4-71c4a32c57fd-9&btsid=0bb0623f16119252544065155e4078&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

not always it is possible to use MAX13487 with the automatic flow control so please make the hardware flow control possible please

michelebergo, 2021-03-28:
> **Describe the problem you have/What new integration you would like**
> Some devices require flow control when communicating via UART. In this case, it will be useful to add the ability to configure [...]

Hi awasaw, can you helping me on implementing the hardware flow control as you did please?

shawly, 2022-01-11:
I looked around and found that the [ESP-IDF framework supports it](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/uart.html) so I extended the uart component to enable hardware flow control for ESP32 boards. It's not yet fully complete since I still need to improve validation a little but it's fully usable I tested it successfully.

https://github.com/shawly/esphome-components

Technically the framework would also support DSR and DTS but I didn't implement it since I can't test it.

ssieb, 2022-10-24:
I don't see why a possible external component is a reason to close a valid feature request.