esphome/feature-requests #1127: Support for all INA2xx devices.

daned33, 2021-02-08:
**Describe the problem you have/What new integration you would like**

Add support for the INA2xx library that has multiple INA2xx High-Side/Low-Side Bi-Directional I2C Current and Power Monitors at the same time.
https://github.com/Zanduino/INA

**Please describe your use case for this integration and alternatives you've tried:**

Currently, there is only support for [INA219](https://esphome.io/components/sensor/ina219.html).
There is a wide range of current sensing chips that would be very convenient to have. Especially if low side current sensing or bi-directional sensing is needed,

**Additional context**

The INA Libary by Zanduino is a great resource for integrating with esphome.

I will be willing to test.

reaper7, 2021-02-17:
@daned33 - INA226 and INA3221 are also supported, check dev branch.
I personally use ina226 and it works very well with esphome dev

daned33, 2021-02-18:
@reaper7 cheers for that.seems like INA220 works off the same as INA219. but I need to test once I get a module.

nagyrobi, 2022-07-06:
Implemented in:
https://esphome.io/components/sensor/ina226.html
https://esphome.io/components/sensor/ina219.html
https://esphome.io/components/sensor/ina3221.html

closed 2022-07-06 by nagyrobi