esphome/feature-requests #1131: Create a virtual/soft cct feature to simulate RGBWW/RGB-CCT with just RGBW

mew1033, 2021-02-15:
**Describe the problem you have/What new integration you would like**
Basically this feature of Tasmota: https://tasmota.github.io/docs/Lights/#virtual-ct. I just got some Teckin SB60 bulbs, and I set them up with Tasmota first, then moved to ESPHome. While I was using Tasmota, I enabled the Soft-CCT feature as per here: https://templates.blakadder.com/teckin_SB60.html#soft-cct-setup. It worked quite well, all things considered. It sounds like tuya was doing it as well. Either way, it'd be awesome if ESPHome could do something similar.

**Please describe your use case for this integration and alternatives you've tried:**
I'd like to simulate having both a warm white and cool white LED, even though I only have one. The other RGB LEDs should be able to help.

ryan-ronnander, 2024-02-11:
My initial journey with ESPHome has led me here. I have been on a mission to get SYLVANIA Smart+ WiFi bulbs working with full local control and free from Tuya's cloud (particularly LEDVANCE/SYLVANIA's rebranded Tuya app/cloud). 

These bulbs are Tuya based using the ([WB8P](https://developer.tuya.com/en/docs/iot/wb8p-module-datasheet?id=K9fwx4f89tvzd)) and work great with Home Assistant using either tuya local or local tuya. They have a CRI > 90, color temp adjustable from 2700-6500K, they're from a name brand manufacturer with a 2-year warranty, and are only roughly $5 per bulb (4-pack prices). 

It wasn't apparent until ripping one of them open (to pave the way for [tuya-cloudcutter](https://github.com/tuya-cloudcutter/tuya-cloudcutter.github.io/pull/88)) that these bulbs were technically RGBW bulbs the whole time:

[image]

The WBP8/BK7231T disqualifies Tasmota, leaving two options for custom firmware:

- ESPHome
- OpenBeken

Tasmota has its `Virtual CT` option, OpenBeken has `Flag 24` - _[LED] Emulate Cool White with RGB in device with four PWMS_ (not perfect with these particular bulbs, but it "works"). From what I can see there have been some attempts to bring similar functionality into ESPHome with some attempts made, but nothing has been merged into the codebase.

I have no idea how many other manufacturers are doing this with RGBW bulbs in circulation. I just wanted to post this here as an example of a manufacturer using emulated color temperature in their firmware with an RGBW bulb. Getting this to work in ESPHome is more about getting full 100% stock functionality after flashing to ESPHome versus giving every RGBW light a bonus/hidden feature.

Hunter32R, 2024-12-07:
I used something like this in esphome, it even works. It would be nice if someone wrote a custom light component.
```
bp5758d:
  data_pin: P6
  clock_pin: P9

output:
  - platform: bp5758d
    id: blue
    channel: 1 # Valid values 1-5 to match OUT1-OUT5
    current: 12 #14 # In mA, valid values 0-90, default 10
  - platform: bp5758d
    id: red
    channel: 2 # Valid values 1-5 to match OUT1-OUT5
    current: 12 #14 # In mA, valid values 0-90, default 10
  - platform: bp5758d
    id: green
    channel: 3 # Valid values 1-5 to match OUT1-OUT5
    current: 12 #14 # In mA, valid values 0-90, default 10
  - platform: bp5758d
    id: warm
    channel: 4 # Valid values 1-5 to match OUT1-OUT5
    current: 30 #40 # In mA, valid values 0-90, default 10
  - platform: bp5758d
    id: cold
    channel: 5 # Valid values 1-5 to match OUT1-OUT5
    current: 0 # In mA, valid values 0-90, default 10
  - platform: template
    id: color_t
    type: float
    write_action:
      - sensor.template.publish:
          id: color_r_s
          state: !lambda return state;
      - sensor.template.publish:
          id: color_g_s
          state: !lambda return state;
      - sensor.template.publish:
          id: color_b_s
          state: !lambda return state;
      - sensor.template.publish:
          id: color_w_s
          state: !lambda return state;
  - platform: template
    id: brightness_t
    type: float
    write_action:
      - sensor.template.publish:
          id: brightness_s
          state: !lambda return state;
  - platform: template
    id: red_t
    type: float
    write_action:
      - sensor.template.publish:
          id: red_s
          state: !lambda return state;
  - platform: template
    id: green_t
    type: float
    write_action:
      - sensor.template.publish:
          id: green_s
          state: !lambda return state;
  - platform: template
    id: blue_t
    type: float
    write_action:
      - sensor.template.publish:
          id: blue_s
          state: !lambda return state;

sensor:
  - platform: template
    name: "color_r"
    internal: True
    id: color_r_s
    update_interval: never
    filters:
      - calibrate_linear:
         method: exact
         datapoints:
          - 0 -> 0          #6500
          - 0.111 -> 0      #6000
          - 0.222 -> 0      #5700
          - 0.333 -> 0.045  #5000
          - 0.444 -> 0.088  #4500
          - 0.556 -> 0.176  #4000
          - 0.667 -> 0.106  #3500
          - 0.822 -> 0      #3000
          - 0.889 -> 0.15   #2500
          - 1 -> 0.3        #2000
  - platform: template
    name: "color_g"
    internal: True
    id: color_g_s
    update_interval: never
    filters:
      - calibrate_linear:
         method: exact
         datapoints:
          - 0 -> 0.675      #6500
          - 0.111 -> 0.66   #6000
          - 0.222 -> 0.47   #5700
          - 0.333 -> 0.563  #5000
          - 0.444 -> 0.554  #4500
          - 0.556 -> 0.5    #4000
          - 0.667 -> 0.285  #3500
          - 0.822 -> 0      #3000
          - 0.889 -> 0      #2500
          - 1 -> 0          #2000
  - platform: template
    name: "color_b"
    internal: True
    id: color_b_s
    update_interval: never
    filters:
      - calibrate_linear:
         method: exact
         datapoints:
          - 0 -> 0.655      #6500
          - 0.111 -> 0.5    #6000
          - 0.222 -> 0.42   #5700
          - 0.333 -> 0.38   #5000
          - 0.444 -> 0.25   #4500
          - 0.556 -> 0.14   #4000
          - 0.667 -> 0.074  #3500
          - 0.822 -> 0      #3000
          - 0.889 -> 0      #2500
          - 1 -> 0          #2000
  - platform: template
    name: "color_w"
    internal: True
    id: color_w_s
    update_interval: never
    filters:
      - calibrate_linear:
         method: exact
         datapoints:
          - 0 -> 0.52       #6500
          - 0.111 -> 0.58   #6000
          - 0.222 -> 0.59   #5700
          - 0.333 -> 0.64   #5000
          - 0.444 -> 0.71   #4500
          - 0.556 -> 0.7    #4000
          - 0.667 -> 0.85   #3500
          - 0.822 -> 1      #3000
          - 0.889 -> 0.93   #2500
          - 1 -> 0.85       #2000
    on_value:
      - output.set_level:
          id: warm
          level: !lambda return x * id(brightness_s).state;
  - platform: template
    name: "brightness"
    internal: True
    id: brightness_s
    update_interval: never
  - platform: template
    name: "red"
    internal: True
    id: red_s
    update_interval: never
    on_value:
      - output.set_level:
          id: red
          level: !lambda return x + id(color_r_s).state * id(brightness_s).state;
  - platform: template
    name: "green"
    internal: True
    id: green_s
    update_interval: never
    on_value:
      - output.set_level:
          id: green
          level: !lambda return x + id(color_g_s).state * id(brightness_s).state;
  - platform: template
    name: "blue"
    internal: True
    id: blue_s
    update_interval: never
    on_value:
      - output.set_level:
          id: blue
          level: !lambda return x + id(color_b_s).state * id(brightness_s).state;

  - platform: wifi_signal
    name: "WiFi Signal"
    update_interval: 60s

light:
  - platform: rgbct
    name: "Light"
    red: red_t
    green: green_t
    blue: blue_t
    color_temperature: color_t
    white_brightness: brightness_t
    cold_white_color_temperature: 6536 K
    warm_white_color_temperature: 2000 K
    restore_mode: ALWAYS_OFF
    color_interlock: True
    gamma_correct: 1.3
    default_transition_length: 1s
```