esphome/feature-requests #1132: Add Humidity sensor in Climate component

manju-rn, 2021-02-15:
**Describe the problem you have/What new integration you would like**

Currently the climate component does not have any humidity information displayed. However, most of the temprature sensors do come with humidity information as well. Just like the sensor for the temp, it would be great if the Humidity can also be included
**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

fabianschaefer, 2021-03-10:
In addition it would be handy to use the climate component for humidity as well. Instead of heating/cooling having humidifying/drying and also show the humidity with % instead of °C in the UI.
I know this can already be achieved by misusing the heating/cooling components, but that looks a bit odd in the frontend.

MiguelAngelLV, 2021-08-21:
In Home Assistant, climate component support current_humidity  (https://developers.home-assistant.io/docs/core/entity/climate/) , so, EspHome must add to climate component.

closed 2022-09-17 by manju-rn