esphome/feature-requests #1133: Only upgrade PlayformIO on run/compile if needed.

lanrat, 2021-02-15:
**Describe the problem you have/What new integration you would like**

When running the esphome cli (in docker) with the compile or run commands, PlatformIO is re-downloaded every time.
This is the slowest part of the command, and takes consumes unnecessary time/bandwith/resources.

**Please describe your use case for this integration and alternatives you've tried:**

If the current version of PlatformIO is already installed, use that instead of re-downloading it every time.

**Additional context**

When developing and refreshing often, this slows down development work a lot.

jesserockz, 2021-02-21:
If you are running it like in the docs:
```
docker run --rm -v "${PWD}":/config -it esphome/esphome livingroom.yaml compile
```
then the platformio "install" is only temporary as the container is created and destroyed everytime you run the command

Try running a container with a shell and then execute the run/compile commands inside that:
```
docker run --rm -v "${PWD}":/config -it --entrypoint bash esphome/esphome
esphome livingroom.yaml run
```

lanrat, 2021-02-21:
> ```
> docker run --rm -v "${PWD}":/config -it --entrypoint bash esphome/esphome
> esphome livingroom.yaml run
> ```

I think this would work.
I guess what I'm proposing is to allow the platformio data that is installed to be installed to a volume that can be made persistent so that it can be reused between docker runs.

jesserockz, 2021-02-21:
We would not want to incorporate that into the main image because it will persist across versions and might end up breaking something, but you could create another volume mount via the one-liner to point to the `~/.platformio` folder inside the container.

lanrat, 2021-02-21:
Thanks, I'll give that a try.

To make sure I don't fall into the same trap you mentioned about persisting across versions, does the existing esphome code check the local platformio version before running? And upgrade if necessary?

jesserockz, 2021-02-22:
the `platformio` command itself is updated with esphome, but whether or not `platformio` actually updates, its up to `platformio`

lanrat, 2021-02-22:
Great!

I added `-v "${DIR}/.platformio/":/root/.platformio/` to my build script's docker command and now my build times are ~5s faster!

closed 2021-02-22 by lanrat