esphome/feature-requests #1134: Add Denon IR Protocol

DelusionalAI, 2021-02-16:
**Describe the problem you have/What new integration you would like**

Add support for the Denon remote like [here](https://forum.arduino.cc/index.php?topic=11649.0) or [tasmota](https://forum.arduino.cc/index.php?topic=11649.0) 

**Please describe your use case for this integration and alternatives you've tried:**

I've tried capurturing the codes raw and replaying but havnet had any luck. I got it first try on Tasmota 9.2 sending. a comand like the below over MQTT.

{
  "Protocol": "DENON",
  "Bits": 15,
  "Data": "0x1B48",
  "DataLSB": "0xD812",
  "Repeat": 0
}

**Additional context**

subcontact, 2021-09-09:
This would be great. Is there any view on how we could get this done?

djansen1987, 2022-04-10:
any luck getting it working with RAW ? Already spending 2 days on figuring it out but not luck. Would be great to implement this!

djansen1987, 2022-04-10:
@agners @oxan @kbx81 @jesserockz 

By looking in the [source code](https://github.com/esphome/esphome/tree/80d03a631e3bccc055d76cc642a6f8993f141df4/esphome/components/remote_base) i see you have worked on some of the previous modules. 

With the information on[ this page](https://forum.arduino.cc/t/denon-ir-remote-decoded/11714) give by @DelusionalAI i think it is a matter of copy/pasting some code around for you guys. 

Are you willing to put some time in this ? Much appreciated, thanks in advanced !!!

andipilz, 2024-01-18:
I got this now working via a Comment in the Sources of Tasmota giving a link to this file: http://assets.denon.com/documentmaster/us/denon%20master%20ir%20hex.xls
It contains for all the models of Denon, which I have, the Pronto codes. And they work!!!

djansen1987, 2024-01-18:
Wow, forget all about this one. This is some great list of codes. Will have a look at this again. Mine is still working via a separate raspberry pi which Is way overkill and I want to control it via an ESP

andipilz, 2024-01-18:
Great! Glad it help not only me.
But I guess the Denon variety of Remotes is so big, it might need some further selection of the right models?!
Would love to help, but not my real programming turf. Gabby to (beta) test, though.