esphome/feature-requests #1135: Specify a DHCP client name (Wifi component)

nalt, 2021-02-16:
**Describe the problem you have/What new integration you would like**

ESPHome does not set a client name when it sends a DHCP request. The name (or a custom config option) should be sent in the DHCP request.

**Please describe your use case for this integration and alternatives you've tried:**

The client name is very helpful: You see in the access point which devices are connected. And you can use the hostname of the device to communicate with it instead of an IP.

**Additional context**

–

randybb, 2021-02-16:
Just checking client list on my Unifi controller and everything seems to be fine. But it is possible that it is more sophisticated than i thought and it uses zeroconf / avahi to discover hostnames.

nalt, 2021-02-16:
OpenWrt does not show a hostname. For most other devices, it does.