esphome/feature-requests #1137: Docker container: add "vi" editor

chemelli74, 2021-02-18:
**Describe the problem you have/What new integration you would like**

Consider adding `vi` to docker image. 
Will not take much space and would be handy to edit yaml file directly there.

**Please describe your use case for this integration and alternatives you've tried:**

Edit the file on the host and share the folder with the container

SenexCrenshaw, 2021-02-19:
I agree!

dougle03, 2021-02-20:
Vi or Nano pls

legomind, 2021-03-18:
If you want to use vi in docker, I would suggest using a shared volume and the vim docker image. 

Otherwise, mount a host folder inside the esphome image, and use vim on your host machine directly.

chemelli74, 2021-03-18:
Why not considering including it by default in esphome container ?

Simone

legomind, 2021-03-18:
Because not only does it go against the container philosophy, there are some of us who do not wish to download larger images with each update.

chemelli74, 2021-03-18:
It's a pity but I got your point.

Closing

closed 2021-03-18 by chemelli74