esphome/feature-requests #1138: Reverse LEDs of light component

Probastian, 2021-02-19:
**Describe the problem you have/What new integration you would like**

Referring to the light component. I would like a feature that reverses the led order passed to the led strip, preferably as an configuration variable (For example this: `reversed: true`, defaults to false).
For further explanation & example see Additional Context.

**Please describe your use case for this integration and alternatives you've tried:**

I've tried making a clone of neopixelbus (The platform I'm using) as a custom component and modifying it, but without success.

**Additional context**

Example: Light Component with LED Length of 10 and is reversed. If I turn LED 1 green, LED 10 on the Light Strip should turn green. If I turn LED 2 green, LED 9 on the Light Strip should turn green. In pseudo-code instead of `turnGreen(ledstrip[index])` it should be `turnGreen(ledstrip[ledstrip.length - index])`

oxan, 2021-06-04:
With esphome/esphome#1484 merged, you can emulate this by using a light partition with a single segment (starting from 1.19):
```
light:
  - platform: partition
    name: "reversed light strip"
    segments:
      - id: light1
        from: 0
        to: 20
        reversed: true
  - platform: fastled_clockelss
    id: light1
    internal: true
    # other settings
```

Probastian, 2022-03-25:
Thanks, closing this.
Good job tho.

closed 2022-03-25 by Probastian