esphome/feature-requests #1139: urinal control with 2 x ds18b20

sargus123, 2021-02-20:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

sargus123, 2021-02-20:
Hello. On espeas, I could turn on the rule for the esp to check whether the temperature of the thermometer (1) in the siphon has changed in relation to the temperature in the second sensor (2) outside the siphon.
I did not find how to write such a rule that esp would analyze temperature differences by itself and on this basis it would execute a sentence independently of HA.

if temperature 1 is higher than temperature 2
then open the flush valve.
if the temperature drops, close the valve

so far I found a comparison of the temperature with one constant and here instead of the constant there is a second thermometer on the basis of which it analyzes whether someone has used the urinal.

jesserockz, 2021-02-21:
Try something like this:

```
sensor:
- name: temp1
  on_value:
    then:
      - if:
          condition:
            lambda: 'return x > id(temp2).state;'
          then:
            - # open the valve
            - wait_until:
                lambda: 'return id(temp1).state < id(temp2).state;'
              then:
                - # close the valve
```