esphome/feature-requests #1142: more pre-defined colors, please

wjcarpenter, 2021-02-22:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**
Now that esphome is directly supporting various color displays and has a fine Color class in C++, it would be nice to have additional pre-defined colors so that folks didn't have to define their own. We have COLOR_WHITE and COLOR_BLACK, but that's not nearly enough for color displays.

**Additional context**

I made a bunch of color definitions that are easy to include in a YAML file. I don't know if this is the best mechanism, but something with the same sort of ease-of-use in esphome itself would be quite nice.

https://community.home-assistant.io/t/here-are-some-color-definitions-for-use-with-esphome/278124
https://gitlab.com/wjcarpenter/esphome-colors

wjcarpenter, 2021-02-23:
Or, if it's helpful, put a pointer in the docs, maybe here: https://esphome.io/guides/configuration-types.html?highlight=color#color , to my colors repo mentioned above. I'd be happy to leave it in place indefinitely.

nagyrobi, 2022-07-07:
How about submitting this as a PR?

closed 2022-07-07 by nagyrobi