esphome/feature-requests #1143: ESPHome write measurements to InfluxDB

michaelpiron, 2021-02-22:
**Describe the problem you have/What new integration you would like**

Currently I use ESPHome on my ESP-based home IoT devices, in combination with Home Assistant. Home Assistant collects the measured data of my ESPHome devices and then sends it to my InfluxDB database.
However, I think it would be useful to let ESPHome devices write their measurements directly into the InfluxDB, without the need of having a running Home Assistant instance.

**Please describe your use case for this integration and alternatives you've tried:**

My use case: I would like to perform power measurements at a friend's house. There is no need for having a Home assistant server. With the proposed feature (ESPHome to InfuxDB) I could have smart plugs with ESPHome firmware connecting directly to a remote/cloud InfluxDB, making it very simple and easy to perform and collect measurements.

This is inspired by the use case of the Iotawatt power monitoring system: it's an ESP-based device, running its specific firmware. This firmware allows to specify the Emoncms/InfluxDB/PVoutput server to which the device should send its data.
Documentation: https://docs.iotawatt.com/en/02_05_11/influxDB.html

It would be cool to have ESPHome based power plugs having similar features: sending their measured data to an InfluxDB.
**Additional context**

In the ESPHome config YAML file it should be possible to specify parameters like:

- Post interval: Number of seconds that each data point will represent.
- Bulk send: Number of interval postings to aggregate into a single HTTP request.
- server URL: URL of the influxDB server.(incl port number)
- database: Name of the influxDB database that should receive the data
- retention policy: Optional name of the influxDB retention policy that you want to associate with the measurements that are written to influxDB
- username/password: Optional security credentials
- measurement: Name that you assign to the measurements that ESPHome will be posting
- field-key: Each measurement contains a field key and field value. If not specified, the default field key in each measurement is the string ‘value’.
- tag-set: A collection of optional user specified tag-key/tag-value pairs that will be included as part of each measurement, for easy querying in InfluxDB.
- measurements: The set of *measurement*s that will be sent at each interval. (voltage, current, power, uptime, etc)

pasiz, 2021-03-08:
https://esphome.io/components/http_request.html

jabastien, 2021-03-10:
a solution that uses http_request can be found here: [https://github.com/jabastien/esphome-influxdb](url)

closed 2022-07-06 by nagyrobi