esphome/feature-requests #1144: What would be the best way to implement a WordClock component

engelant, 2021-02-24:
**Describe the problem you have/What new integration you would like**
I would like to create a custom effect utilizing addressable RGBW LEDs to control a [WordClock](https://www.instructables.com/Und-noch-eine-Wordclock/).

**Please describe your use case for this integration and alternatives you've tried:**
Currently there are a few solutions out there, but esphome seems to be one of the more wholesome solutions (OTA, fallback WiFi, HomeAssistant Integration, Software current Limit for LEDs, etc...). 

**Additional context**
Basically the code is quiet easy, but It propably would be the easiest to write in C++ (Word/LED mapping etc.).

From my current understanding the way to go would be to use labda effects.
On the other hand, the layout of a WordClock is different than a normal addressable stripe (zig zack grid, depends on implementation). Would it make more sense to write it as a custom light output? Also there are advanced design options (like backgroud/foreground color), that idealy should be configurable trough HomeAssistant (don't know if that's even possible).
Further refresh intervals would need to be bound to NTP time (or any other time source) to refresh it every 60s, but tied to every full minute.

I would be very thankful for a recomendation of the cleanest way to implement such a functionality and which components to utilize from esphome.

noxhirsch, 2021-03-19:
Have you made any progress in this regard? I'm looking for a solution for my wordclock, too.

engelant, 2021-03-19:
Unfortunatly, as you can see, I didn't get anyfeedback. It's not that urgent on my side, so I didn't want to dig in too deep, but it seems at some point in the future I'll have to go deeper...

noxhirsch, 2021-03-20:
I had a faint hope that you found a solution somewhere else ;) 
It's not that urgent on my side, too - since my Wordclock currently runs with a simple "Arduino" code. Maybe someone else has a great idea, otherwise I'll do it myself sometime when I have more time.

nagyrobi, 2022-07-01:
Please join our Discord server for more prompt answers to such concrete questions.

closed 2022-07-01 by nagyrobi