esphome/feature-requests #1151: Support for the CSE7761 (as found in the Sonoff DualR3)

HA-TB303, 2021-03-04:
**Describe the problem you have/What new integration you would like**
Please add support for the CSE7761 power metering

**Please describe your use case for this integration and alternatives you've tried:**
CSE7766 is supported, but the Sonoff DualR3 uses CSE7761

**Additional context**
More info:
https://github.com/kakopappa/CSE7761
http://www.chipsea.com/gongsixinwen/jishu--zhinengdianqiloudianbaohuzheyangshejigengka.html
https://github.com/arendst/Tasmota/discussions/10793

HA-TB303, 2021-03-18:
It's done in tasmota: https://github.com/arendst/Tasmota/blob/development/tasmota/xnrg_19_cse7761.ino#L24

Drew765, 2021-04-27:
Seconding support. Happy to assist with testing/logging requirements.

lednicazar, 2021-04-29:
Will be great, having the DualR3 working in esphome with the power meter will be an awesome thing

alimnette, 2021-06-25:
hello, i need this feature too. 👍🏼

tazginpax-ho, 2021-08-15:
I would really love to get this in ESP Home rather than Tasmota

HA-TB303, 2021-08-15:
@cdheiser, why is this **### feature request** closed and pointed to just another issues (which is also closed)?

alimnette, 2021-08-16:
Same question? i need support of CSE7761.

Can you reopen this request plz?

MattXcz, 2021-08-18:
It would help a lot... thanks @OttoWinter

MattXcz, 2021-08-18:
Is it also posible to add Sonoff DUALR3 temperature reading

Drew765, 2021-08-18:
Where is the temp sensor?

MattXcz, 2021-08-19:
I dont know, but it shows CPU temperature in Tasmota...maybe it is part of CSE7761 ?

wouterbouvy, 2021-08-19:
> I dont know, but it shows CPU temperature in Tasmota...maybe it is part of CSE7761 ?

It's part of the ESP32 chip, but feature request to read this temperature in ESPHome have not been picked up since there are different versions of the ESP32, some without temperature sensor.

MattXcz, 2021-08-19:
Ok, understood. Maybe there can be Sonoff DualD3 as board with this feauture. For me this temerature readings are important, since i plan runnig this in the atique...

SvbZ3r0, 2021-09-21:
I've got a spare and would like to help bring full support for the device. What can I do?

berfenger, 2021-09-27:
I just published an external component to support the CSE7761 serial sensor.
Here is an example: https://github.com/berfenger/esphome_components/blob/main/example_sonoff_dual_r3_v1.yaml

Let me know if it works for you because I'm planning to issue a merge request to the esphome mainline.

eduperez, 2021-10-02:
@berfenger I tried your component, the entities show on HA, but the state is stuck on "unknown" for all of them. Perhaps my device is an incompatible version; how can I know which version of the device do I own? Thanks!

berfenger, 2021-10-02:
@eduperez You can check the hardware version on the back side of the PCB. It should have a text like "DUALR3-POW V1.6". V2 has a different chip and does not work with this component.

MattXcz, 2021-10-03:
For me, it works perfectly :) Thank you

eduperez, 2021-10-03:
@berfenger The devices are already installed, and in a hard to reach place, can't this be checked remotely?

SvbZ3r0, 2021-10-03:
> I just published an external component to support the CSE7761 serial sensor. Here is an example: https://github.com/berfenger/esphome_components/blob/main/example_sonoff_dual_r3_v1.yaml
> 
> Let me know if it works for you because I'm planning to issue a merge request to the esphome mainline.

Works perfectly for me. 
I even cross checked the voltage readings with my multimeter, just to be sure.

I'm guessing your driver is good to go for the mainline PR.

berfenger, 2021-10-04:
> @berfenger The devices are already installed, and in a hard to reach place, can't this be checked remotely?

I'm afraid it cannot be checked via software. The only difference between v1.x and v2.0 is the power consumption sensor, so I would assume your devices are v2.0.

I will try to buy a device from the sonoff official store hoping they send me an updated version and implement a driver/component for the new sensor. But I can't promise anything.

Drew765, 2021-10-04:
The v2.x is merged in Tasmota already I think. You could use that for the time being maybe? I wish I could help, but I've just started learning this stuff so I can contribute in future.

skarcha, 2021-10-10:
Works perfect for me as well. It's running fine for five days without problems.

Thanks!